openwrt/asu #1211: Attended Sysupgrade fails from 23.05.5 to 24.10 on GL.iNet MT6000

Antares2619, 2025-02-07:
**Request Data:**
```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24106-10cc5fcd00",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "glinet,gl-mt6000",
    "target": "mediatek/filogic",
    "version": "24.10.0",
    "packages": [
        "attendedsysupgrade-common",
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "collectd-mod-thermal",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "eip197-mini-firmware",
        "f2fsck",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iperf3",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-authenc",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-des",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-hw-safexcel",
        "kmod-crypto-manager",
        "kmod-crypto-md5",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha1",
        "kmod-crypto-sha256",
        "kmod-crypto-sha512",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-libphy",
        "kmod-mac80211",
        "kmod-mt76-connac",
        "kmod-mt76-core",
[... 182 lines cut ...]
 * pkg_hash_check_unresolved: cannot find dependency collectd-mod-cpu for luci-app-statistics
 * pkg_hash_check_unresolved: cannot find dependency collectd-mod-memory for luci-app-statistics
 * pkg_hash_check_unresolved: cannot find dependency collectd-mod-interface for luci-app-statistics
 * pkg_hash_check_unresolved: cannot find dependency collectd-mod-load for luci-app-statistics
 * pkg_hash_check_unresolved: cannot find dependency collectd-mod-network for luci-app-statistics
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-statistics found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-statistics.
 * pkg_hash_check_unresolved: cannot find dependency ttyd for luci-app-ttyd
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-ttyd found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-ttyd.
 * pkg_hash_check_unresolved: cannot find dependency watchcat for luci-app-watchcat
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-app-watchcat found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-app-watchcat.
 * pkg_hash_check_unresolved: cannot find dependency modemmanager for luci-proto-modemmanager
 * pkg_hash_fetch_best_installation_candidate: Packages for luci-proto-modemmanager found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package luci-proto-modemmanager.
 * opkg_install_cmd: Cannot install package picocom.
 * opkg_install_cmd: Cannot install package python3.
 * opkg_install_cmd: Cannot install package speedtestcpp.
 * opkg_install_cmd: Cannot install package watchcat.
 * opkg_install_cmd: Cannot install package owut.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2
```

lewi3069, 2025-02-07:
Had this on my MT6000 as well.

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).