openwrt/asu #1220: Error building the firmware image: Selected packages exceed device storage

andriej, 2025-02-08:
Except that until previous -rc's the space was enough.
With -rc5 it's 43% of /overlay (38 of 73MB taken)

Nanopi R5C

```
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28304-6dacba30a7",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "3ef4ebd33d54bb6e2c321efc1c1abaf07c88d6e37749c2c55ae5daea79ac497a",
    "sha256_unsigned": "",
    "client": "luci/25.037.68331~42f464c",
    "packages": {
        "kmod-usb-storage": "6.6.69-r1",
        "luci-proto-wireguard": "25.005.74981~65c86ed",
        "libopenssl": "3.0.15-r1",
        "openssh-sftp-server": "9.9_p1-r1",
        "iputils-arping": "20240905-r1",
        "libc": "1.2.5-r4",
        "mkf2fs": "1.16.0-r3",
        "opkg": "2024.10.16~38eccbb1-r1",
        "kmod-usb-core": "6.6.69-r1",
        "luci-app-vnstat2": "25.005.74981~65c86ed",
        "diffutils": "3.10-r1",
        "kmod-tun": "6.6.69-r1",
        "openssh-server": "9.9_p1-r1",
        "kmod-rtw88-8822ce": "6.6.69.6.12.6-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1649~6dacba30a7",
        "mosquitto-client-ssl": "2.0.18-r3",
        "partx-utils": "2.40.2-r1",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "6in4": "29",
        "kmod-r8169": "6.6.69-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "uboot-envtools": "2024.07-r1",
        "dnsmasq": "2.90-r3",
        "usbutils": "017-r1",
        "kmod-usb-net-dm9601-ether": "6.6.69-r1",
        "exfat-fsck": "1.2.5-r1",
        "mdns-repeater": "2020.05.03~921d8850-r2",
        "etherwake": "1.09-r5",
        "usbids": "0.387-r1",
        "block-mount": "2024.07.14~408c2cc4-r1",
        "kmod-fs-vfat": "6.6.69-r1",
        "kmod-usb2": "6.6.69-r1",
        "kmod-usb-serial-option": "6.6.69-r1",
        "kmod-fs-exfat": "6.6.69-r1",
        "cJSON": "1.7.17-r1",
        "tcpdump": "4.99.5-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "iptables-mod-ipopt": "1.8.10-r1",
        "uci": "2023.08.10~5781664d-r1",
        "nmap-full": "7.95-r1",
        "luci-ssl": "25.005.74981~65c86ed",
        "dropbear": "2024.86-r1",
        "tailscale": "1.78.1-r1",
        "curl": "8.10.1-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "smstools3": "3.1.21-r4",
        "luci-app-sqm": "25.005.74981~65c86ed",
        "urandom-seed": "3",
        "ppp": "2.5.1-r1",
        "arp-scan-database": "1.10.0-r2",
        "bcp38": "5-r11",
        "kmod-gpio-button-hotplug": "6.6.69-r5",
[... 1026 lines cut ...]
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/aarch64_generic/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Pseudo file "dev" exists in source filesystem "/builder/build_dir/target-aarch64_generic_musl/root-rockchip/dev".
Ignoring, exclude it (-e/-ef) to override.
error: ext4_allocate_best_fit_partial: failed to allocate 2 blocks, out of space?
make[3]: *** [/builder/include/image.mk:399: /builder/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/root.ext4] Error 1
make[2]: *** [Makefile:268: build_image] Error 2
make[1]: *** [Makefile:163: _call_image] Error 2
```

andriej, 2025-02-11:
Turned out to be problem with the storage assigned to ASU.
After removing tcpdump and nmap I was able to get the image. It might be worth to assign a little more resources to ASU tho ;-)

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).

andriej, 2025-02-15:
Confirmed, issue is solved and image is built without issue.

closed 2025-02-15 by andriej