openwrt/asu #1226: Error building the firmware image Server response: Error: Impossible package selection

mrobo11, 2025-02-08:
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "cee6febac1f3c3dc0b824cebcdf9385538e44da64694a23917aef0604257dac5",
    "sha256_unsigned": "",
    "client": "luci/25.037.68331~42f464c",
    "packages": {
        "kmod-usb-storage": "6.6.73-r1",
        "luci-app-statistics": "25.037.68331~42f464c",
        "luci-app-watchcat": "25.037.68331~42f464c",
        "luci-app-nlbwmon": "25.037.68331~42f464c",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "kmod-usb-core": "6.6.73-r1",
        "kmod-bonding": "6.6.73-r1",
        "kmod-usb-storage-uas": "6.6.73-r1",
        "luci-app-vnstat2": "25.037.68331~42f464c",
        "libpthread": "1.2.5-r4",
        "mjpg-streamer-www-simple": "1.0.0-r6",
        "rpcd": "2024.09.17~9f4b86e7-r1",
        "kmod-nft-fib": "6.6.73-r1",
        "kmod-nfnetlink": "6.6.73-r1",
        "kmod-ata-ahci-platform": "6.6.73-r1",
        "kmod-nf-reject6": "6.6.73-r1",
        "kmod-usb-dwc3": "6.6.73-r1",
        "luci-mod-system": "25.037.68331~42f464c",
        "kmod-nf-flow": "6.6.73-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "px5g-mbedtls": "11",
        "getrandom": "2024.04.26~85f10530-r1",
        "luci-app-lldpd": "25.037.68331~42f464c",
        "luci-theme-bootstrap": "25.037.68331~42f464c",
        "kmod-pppoe": "6.6.73-r1",
        "kmod-video-core": "6.6.73-r1",
        "kmod-pppox": "6.6.73-r1",
        "kmod-nf-reject": "6.6.73-r1",
        "ath10k-firmware-qca9984-ct": "2020.11.08-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "kmod-nf-nat": "6.6.73-r1",
        "kmod-crypto-crc32c": "6.6.73-r1",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "kmod-usb-dwc3-qcom": "6.6.73-r1",
        "luci-app-uhttpd": "25.037.68331~42f464c",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "uboot-envtools": "2024.07-r1",
        "kmod-usb-ohci": "6.6.73-r1",
        "zerotier": "1.14.1-r1",
        "dnsmasq": "2.90-r4",
        "usbutils": "017-r1",
        "kmod-video-uvc": "6.6.73-r1",
        "kmod-video-gspca-sn9c20x": "6.6.73-r1",
        "rpcd-mod-ucode": "2024.09.17~9f4b86e7-r1",
        "kmod-crypto-acompress": "6.6.73-r1",
        "ucode-mod-math": "2024.07.22~b610860d-r3",
        "luci-app-advanced-reboot": "1.0.1-r16",
        "kmod-lib-crc32c": "6.6.73-r1",
        "kmod-usb-ehci": "6.6.73-r1",
        "luci-mod-status": "25.037.68331~42f464c",
        "luci-app-natmap": "25.037.68331~42f464c",
        "kmod-nft-nat": "6.6.73-r1",
        "kmod-usb2": "6.6.73-r1",
        "kmod-usb3": "6.6.73-r1",
        "luci-app-firewall": "25.037.68331~42f464c",
        "luci-app-filemanager": "25.037.68331~42f464c",
[... 102 lines cut ...]
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a15_neon-vfpv4/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a15_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a15_neon-vfpv4/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a15_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a15_neon-vfpv4/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a15_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a15_neon-vfpv4/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * check_data_file_clashes: Package mjpg-streamer-www-simple wants to install file /builder/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/www/webcam/index.html
	But that file is already provided by package  * mjpg-streamer-www
 * opkg_install_cmd: Cannot install package mjpg-streamer-www-simple.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).

andybjackson, 2025-02-27:
Check again. I had the same problem which has now resolved itself.

If at all possible, rather add to an existing issue than opening a new one. In this instance there were many issues opened against the same symptoms.