openwrt/asu #1227: Error building the firmware image Server response: Error: Impossible package selection

samiulbasher, 2025-02-09:
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28417-daef29c75d",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.027.83426~170375e",
    "packages": {
        "tcpdump-mini": "4.99.5-r1",
        "luci-app-adblock": "25.027.83426~170375e",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "libpthread": "1.2.5-r4",
        "rpcd": "2024.09.17~9f4b86e7-r1",
        "kmod-btmrvl": "6.6.73-r1",
        "libiwinfo": "2024.10.20~b94f066e-r1",
        "ip-full": "6.11.0-r1",
        "kmod-nft-fib": "6.6.73-r1",
        "kmod-nfnetlink": "6.6.73-r1",
        "kmod-nf-reject6": "6.6.73-r1",
        "nano": "8.3-r1",
        "kmod-nf-flow": "6.6.73-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "px5g-mbedtls": "11",
        "getrandom": "2024.04.26~85f10530-r1",
        "kmod-pppoe": "6.6.73-r1",
        "kmod-pppox": "6.6.73-r1",
        "kmod-nf-reject": "6.6.73-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1653~daef29c75d",
        "kmod-nf-nat": "6.6.73-r1",
        "kmod-crypto-crc32c": "6.6.73-r1",
        "luci-app-ttyd": "25.027.83426~170375e",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "adblock": "4.2.4-r1",
        "uboot-envtools": "2024.07-r1",
        "dnsmasq": "2.90-r4",
        "kmod-crypto-acompress": "6.6.73-r1",
        "rpcd-mod-ucode": "2024.09.17~9f4b86e7-r1",
        "ucode-mod-math": "2024.07.22~b610860d-r3",
        "kmod-lib-crc32c": "6.6.73-r1",
        "kmod-nft-nat": "6.6.73-r1",
        "libmbedtls": "3.6.2-r1",
        "libmnl": "1.0.5-r1",
        "jansson": "2.14-r3",
        "tcpdump": "4.99.5-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uclient-fetch": "2024.10.22~88ae8f20-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "luci-ssl": "24.328.43591~9e70c9a",
        "dropbear": "2024.86-r1",
        "mwlwifi-firmware-88w8964": "2024.04.19~a737d348-r2",
        "libnftnl": "1.2.8-r1",
        "rpcd-mod-file": "2024.09.17~9f4b86e7-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "kmod-mwifiex-sdio": "6.6.73.6.12.6-r1",
        "kmod-nf-log": "6.6.73-r1",
        "urandom-seed": "3",
        "ppp": "2.5.1-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-nf-log6": "6.6.73-r1",
        "openvpn-openssl": "2.6.12-r1",
        "luci-app-attendedsysupgrade": "25.027.83426~170375e",
        "ds-lite": "9",
        "kmod-mwlwifi": "6.6.73.2024.04.19~a737d348-r2",
        "attendedsysupgrade-common": "9",
        "kmod-nft-offload": "6.6.73-r1",
        "kmod-ppp": "6.6.73-r1",
        "uhttpd": "2023.06.25~34a8a74d-r4",
        "kmod-nf-conntrack": "6.6.73-r1",
        "ca-bundle": "20240203-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "luci": "24.297.79519~bcd13b9",
        "kmod-nf-conntrack6": "6.6.73-r1",
        "kmod-lib-lzo": "6.6.73-r1",
        "map": "7",
        "rpcd-mod-iwinfo": "2024.09.17~9f4b86e7-r1",
        "kmod-nft-core": "6.6.73-r1",
        "uhttpd-mod-ubus": "2023.06.25~34a8a74d-r4",
        "urngd": "2023.11.01~44365eb1-r1",
        "kmod-slhc": "6.6.73-r1",
        "ppp-mod-pppoe": "2.5.1-r1",
        "luci-app-openvpn": "25.037.68331~42f464c"
    },
    "profile": "linksys,wrt3200acm",
    "target": "mvebu/cortexa9",
    "version": "24.10.0-rc7",
    "diff_packages": true,
    "filesystem": "squashfs"
}
STDERR:
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/mvebu/cortexa9/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../builder/dl/openwrt_core
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/mvebu/cortexa9/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a9_vfpv3-d16/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../builder/dl/openwrt_base
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a9_vfpv3-d16/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/mvebu/cortexa9/kmods/6.6.73-1-b4e6dc3c3253bf5b438f90392080acc3/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../builder/dl/openwrt_kmods
Downloading http://downloads.openwrt.org/releases/24.10.0/targets/mvebu/cortexa9/kmods/6.6.73-1-b4e6dc3c3253bf5b438f90392080acc3/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a9_vfpv3-d16/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a9_vfpv3-d16/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a9_vfpv3-d16/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a9_vfpv3-d16/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a9_vfpv3-d16/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a9_vfpv3-d16/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a9_vfpv3-d16/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/arm_cortex-a9_vfpv3-d16/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * check_data_file_clashes: Package tcpdump-mini wants to install file /builder/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/usr/bin/tcpdump
	But that file is already provided by package  * tcpdump
 * opkg_install_cmd: Cannot install package tcpdump-mini.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).

samiulbasher, 2025-02-16:
Thank you. i will check

andybjackson, 2025-02-27:
Check again. I had the same problem which has now resolved itself.

If at all possible, rather add to an existing issue than opening a new one. In this instance there were many issues opened against the same symptoms.