openwrt/asu #1229: Error building the firmware image Server response: Error: Impossible package selection

BeBiMa, 2025-02-09:
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24106-10cc5fcd00",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "request_hash": "5c35fd49ee50bfc5344886ba2f8a29c2f90afa7e6e91e0d06d7de37b2baa6b74",
    "profile": "generic",
    "target": "x86/geode",
    "version": "24.10.0",
    "packages": [
        "announce",
        "attendedsysupgrade-common",
        "auc",
        "avahi-daemon-service-http",
        "avahi-daemon-service-ssh",
        "avahi-dbus-daemon",
        "avahi-utils",
        "base-files",
        "boinc",
        "boinc-wrapper",
        "busybox",
        "bzip2",
        "ca-bundle",
        "cgi-io",
        "collectd-mod-df",
        "collectd-mod-disk",
        "collectd-mod-dns",
        "collectd-mod-entropy",
        "collectd-mod-ethstat",
        "collectd-mod-ntpd",
        "collectd-mod-sensors",
        "coreutils",
        "coreutils-stat",
        "curl",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "ethtool-full",
        "fdisk",
        "firewall4",
        "fping",
        "fstools",
        "fwtool",
        "getrandom",
        "grub2",
        "grub2-bios-setup",
        "grub2-editenv",
        "grub2-efi",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-button-hotplug",
        "kmod-crypto-acompress",
        "kmod-crypto-arc4",
        "kmod-crypto-cbc",
        "kmod-crypto-crc32c",
        "kmod-crypto-ecb",
        "kmod-crypto-hash",
        "kmod-crypto-hw-geode",
        "kmod-fs-cifs",
        "kmod-fs-vfat",
        "kmod-input-core",
        "kmod-ledtrig-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-lp",
[... 225 lines cut ...]
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/i386_pentium-mmx/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-i386_pentium-mmx_musl/root-x86/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/i386_pentium-mmx/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/i386_pentium-mmx/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-i386_pentium-mmx_musl/root-x86/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/i386_pentium-mmx/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/i386_pentium-mmx/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-i386_pentium-mmx_musl/root-x86/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/i386_pentium-mmx/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/i386_pentium-mmx/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-i386_pentium-mmx_musl/root-x86/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.0/packages/i386_pentium-mmx/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-i386_pentium-mmx_musl/root-x86/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package libavahi-compat-libdnssd.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).

andybjackson, 2025-02-27:
Check again. I had the same problem which has now resolved itself.

If at all possible, rather add to an existing issue than opening a new one. In this instance there were many issues opened against the same symptoms.