openwrt/asu #1231: Couldn't upgrade Flint 2 to 24.10.0 because of `umdns`

richb-hanover, 2025-02-09:
My GL.iNet MT6000 Flint 2 router was running 23.05.5 with a small set of additional packages (see [config-spare-router.sh](https://github.com/richb-hanover/OpenWrtScripts/blob/main/config-spare-router.sh) line 180 for the full list) including `umdns`. 

I used LuCI Attended Sysupgrade, Advance mode, chose 24.10.0, and clicked OK. I got an error that it could not proceed because of `umdns`. (Sorry, I no longer have the exact report.)

I removed `umdns` by clicking the red X next to it, and re-requested. The ASU server built the binary which flashed as expected.

umdns was not installed (I could not find "hostname.local"). I re-installed umdns from LuCI, and now the name works again.

No particular response needed. I report this only to say that I saw this problem, and found a workaround. Thanks!

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).

richb-hanover, 2025-02-14:
In https://github.com/openwrt/asu/issues/1201#issuecomment-2660432338 I commented that I was able to request an image for my Flint 2 that **did** have `umdns` installed. I believe this can be closed, and attributed the (not fixed) problem with the ASU system

closed 2025-02-14 by richb-hanover