openwrt/asu #1237: Attended sysupgrade Error: Impossible package selection

Coroo1, 2025-02-10:
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24012-d8dd03c46f",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "xiaomi,redmi-router-ax6s",
    "target": "mediatek/mt7622",
    "version": "23.05.5",
    "packages": [
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "collectd-mod-ping",
        "curl",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha512",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-mt76-connac",
        "kmod-mt76-core",
        "kmod-mt7615-common",
        "kmod-mt7615e",
        "kmod-mt7622-firmware",
        "kmod-mt7915-firmware",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nft-tproxy",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "libblobmsg-json",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "liblucihttp",
        "liblucihttp-ucode",
        "libmbedtls",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-sqm",
        "luci-app-ssclash",
        "luci-app-statistics",
        "luci-base",
        "luci-light",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "nano",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-mbedtls",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "rpcd-mod-ucode",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-html",
        "ucode-mod-math",
        "ucode-mod-nl80211",
        "ucode-mod-rtnl",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "ucode-mod-uloop",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-mbedtls"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-25.037.68351-2ca3f19"
}
STDERR:
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://downloads.openwrt.org/releases/23.05.5/targets/mediatek/mt7622/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_core
Downloading http://downloads.openwrt.org/releases/23.05.5/targets/mediatek/mt7622/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_base
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package luci-app-ssclash.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

sigurasg, 2025-02-10:
Same here, auc fails with the same or similar errors.

Request data:
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24106-10cc5fcd00",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "request_hash": "61155e22f0365dd8abb5001eb8f126d68fd159629ee7db4f3a200bc63293ed65",
    "profile": "ubnt,edgerouter-x",
    "target": "ramips/mt7621",
    "version": "23.05.5",
    "packages": [
        "acme-acmesh-dnsapi",
        "auc",
        "avahi-dbus-daemon",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "conntrack",
        "coreutils-sort",
        "ddns-scripts-cloudflare",
        "dnsmasq",
        "dropbear",
        "ethtool",
        "firewall4",
        "fstools",
        "fwtool",
        "gawk",
        "getrandom",
        "grep",
        "hostapd-common",
        "htop",
        "iftop",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-gpio-button-hotplug",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
[... 123 lines cut ...]
 * opkg_install_pkg: Checksum or size mismatch for package libexpat. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package avahi-dbus-daemon.
 * opkg_install_pkg: Checksum or size mismatch for package conntrack. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package conntrack.
 * opkg_install_pkg: Checksum or size mismatch for package coreutils-sort. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package coreutils-sort.
 * opkg_install_pkg: Checksum or size mismatch for package gawk. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package gawk.
 * opkg_install_pkg: Checksum or size mismatch for package acme-common. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package luci-app-acme.
 * opkg_install_pkg: Checksum or size mismatch for package luci-app-adblock-fast. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package luci-app-adblock-fast.
 * opkg_install_pkg: Checksum or size mismatch for package sed. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package sed.
 * opkg_install_pkg: Checksum or size mismatch for package shadow-groupadd. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package shadow-groupadd.
 * opkg_install_pkg: Checksum or size mismatch for package shadow-useradd. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package shadow-useradd.
 * opkg_install_pkg: Checksum or size mismatch for package shadow-usermod. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package shadow-usermod.
 * opkg_install_pkg: Checksum or size mismatch for package sudo. Either the opkg or the package index are corrupt. Try 'opkg update'.
 * opkg_install_cmd: Cannot install package sudo.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).

sigurasg, 2025-02-15:
Works for me now. Thanks!

closed 2025-02-15 by aparcar