openwrt/asu #1246: Error: Impossible package selection

jtruland, 2025-02-13:
Error building the firmware image
Server response: Error: Impossible package selection

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24012-d8dd03c46f",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "request_hash": "b676d7f239c4d8822c3b0d271927a9469681f3c3399031ebade72df9776e3440",
    "profile": "raspberrypi,4-model-b",
    "target": "bcm27xx/bcm2711",
    "version": "23.05.4",
    "packages": [
        "base-files",
        "bash",
        "bcm27xx-gpu-fw",
        "bcm27xx-userland",
        "brcmfmac-nvram-43455-sdio",
        "busybox",
        "ca-bundle",
        "cfdisk",
        "cgi-io",
        "coreutils-base64",
        "coreutils-cat",
        "coreutils-od",
        "curl",
        "cypress-firmware-43455-sdio",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "firewall4",
        "fstools",
        "fwtool",
        "gammu",
        "getrandom",
        "git",
        "gsm-utils",
        "hostapd-common",
        "ip6tables-nft",
        "iptables-nft",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-brcmfmac",
        "kmod-crypto-acompress",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-fs-vfat",
        "kmod-hid",
        "kmod-hid-generic",
        "kmod-input-core",
        "kmod-input-evdev",
        "kmod-ipt-conntrack",
        "kmod-ipt-conntrack-extra",
        "kmod-ipt-conntrack-label",
        "kmod-ipt-nat",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-nls-utf8",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-r8169",
        "kmod-slhc",
        "kmod-sound-arm-bcm2835",
        "kmod-sound-core",
        "kmod-usb-core",
        "kmod-usb-hid",
        "kmod-usb-net-lan78xx",
        "kmod-usb-net-qmi-wwan",
        "kmod-usb-serial-ftdi",
        "kmod-usb-serial-qualcomm",
        "libblkid",
        "libblobmsg-json",
        "libcomerr",
        "libext2fs",
        "libf2fs",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "liblucihttp",
        "liblucihttp-ucode",
        "libmbedtls",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libruby",
        "libsmartcols",
        "libss",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "libuuid",
        "logd",
        "luci",
        "luci-app-3ginfo-lite",
        "luci-app-argon-config",
        "luci-app-atcommands",
        "luci-app-attendedsysupgrade",
        "luci-app-cloudflared",
        "luci-app-diag-core",
        "luci-app-firewall",
        "luci-app-modemband",
        "luci-app-modemdefine",
        "luci-app-opkg",
        "luci-app-sms-tool-js",
        "luci-app-tailscale",
        "luci-base",
        "luci-compat",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-ipkg",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-light",
        "luci-lua-runtime",
        "luci-mod-admin-full",
        "luci-mod-dashboard",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-bonding",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-proto-qmi",
        "luci-ssl",
        "luci-theme-argon",
        "luci-theme-bootstrap",
        "luci-theme-material",
        "luci-theme-openwrt",
        "luci-theme-openwrt-2020",
        "mkf2fs",
        "modemband",
        "mtd",
        "nano-full",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "partx-utils",
        "picocom",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-mbedtls",
        "python3-speedtest-cli",
        "qfirehose",
        "qmi-utils",
        "resize2fs",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "rpcd-mod-ucode",
        "ruby",
        "ruby-base64",
        "shadow-useradd",
        "smstools3",
        "tailscale",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-html",
        "ucode-mod-math",
        "ucode-mod-nl80211",
        "ucode-mod-rtnl",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "ucode-mod-uloop",
        "uhttpd",
        "uhttpd-mod-ubus",
        "uqmi",
        "urandom-seed",
        "usign",
        "wireless-regdb",
        "wpad-basic-mbedtls",
        "znc-mod-push"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-25.037.68351-2ca3f19"
}
STDERR:
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://downloads.openwrt.org/releases/23.05.4/targets/bcm27xx/bcm2711/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_core
Downloading http://downloads.openwrt.org/releases/23.05.4/targets/bcm27xx/bcm2711/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/23.05.4/packages/aarch64_cortex-a72/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_base
Downloading http://downloads.openwrt.org/releases/23.05.4/packages/aarch64_cortex-a72/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/23.05.4/packages/aarch64_cortex-a72/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/23.05.4/packages/aarch64_cortex-a72/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/23.05.4/packages/aarch64_cortex-a72/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/23.05.4/packages/aarch64_cortex-a72/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/23.05.4/packages/aarch64_cortex-a72/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/23.05.4/packages/aarch64_cortex-a72/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/23.05.4/packages/aarch64_cortex-a72/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/23.05.4/packages/aarch64_cortex-a72/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package luci-app-3ginfo-lite.
 * opkg_install_cmd: Cannot install package luci-app-argon-config.
 * opkg_install_cmd: Cannot install package luci-app-atcommands.
 * opkg_install_cmd: Cannot install package luci-app-modemband.
 * opkg_install_cmd: Cannot install package luci-app-modemdefine.
 * opkg_install_cmd: Cannot install package luci-app-sms-tool-js.
 * opkg_install_cmd: Cannot install package luci-app-tailscale.
 * opkg_install_cmd: Cannot install package luci-theme-argon.
 * opkg_install_cmd: Cannot install package modemband.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

Ztrby, 2025-02-14:
i have the same issue ,but other apps in the end. 

Collected errors:

-  opkg_install_pkg: Checksum or size mismatch for package hostapd-common. Either the opkg or the package index are corrupt. Try 'opkg update'.
- opkg_install_cmd: Cannot install package hostapd-common.
- opkg_install_pkg: Checksum or size mismatch for package wifi-scripts. Either the opkg or the package index are corrupt. Try 'opkg update'.
- opkg_install_cmd: Cannot install package kmod-mt7603.
- opkg_install_pkg: Checksum or size mismatch for package wpad-basic-mbedtls. Either the opkg or the package index are corrupt. Try 'opkg update'.
- opkg_install_cmd: Cannot install package wpad-basic-mbedtls.
- opkg_install_pkg: Checksum or size mismatch for package wpad-basic-mbedtls. Either the opkg or the package index are corrupt. Try 'opkg update'.
- opkg_install_cmd: Cannot install package wpad-basic-mbedtls.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).

andybjackson, 2025-02-27:
Check again. I had the same problem which has now resolved itself today for me.

If at all possible, rather add to an existing issue than opening a new one. In this instance there were many issues opened against the same symptoms.