openwrt/asu #1248: auc: 0.3.2-1 on 23.05.4 doesn't offer 24.10.0

opty77, 2025-02-13:
auc 0.3.2-1 on 23.05.4 doesn't offer 24.10.0 at least for `ath79/generic (tplink,tl-wdr3600-v1)` and `ramips/mt7621 (mtc,wr1201)`.

efahl, 2025-02-14:
What parameters did you use with `auc`?  Should look something like this:
```
auc -b 24.10
```

gentrydeng, 2025-02-14:
https://forum.openwrt.org/t/resolved-openwrt-image-builder-errors/224906

> 2025-Feb-14: The issues with the image builder system should now be resolved. If problems persist with Attended Sysupgrade (ASU), OpenWrt Upgrade Tool (OWUT) or the Firmware Selector custom image tool, please report them in [this thread](https://forum.openwrt.org/t/notice-problems-with-custom-firmware-image-creation/224829/4).

opty77, 2025-02-16:
I just tried `auc -ncb 24.10` and `auc -ncB 24.10.0` again and it still offers only 23.05.5.

opty77, 2025-02-19:
Maybe because auc got replaced by owut?

closed 2025-02-26 by opty77