openwrt/asu #1257: [feature request] expand image size option

hadifarnoud, 2025-02-18:
I was hoping to build an expanded image with asu. is it possible?

I tried everything else for expanding the SD. It seems to corrupt the filesystem on reboot and it is a hassle. this would be an awesome option if it does not exist.

asu also throws an error if I put too many packages to install because it runs out of space  (#1255). it would be useful to expand the image size for this reason as well

aparcar, 2025-02-18:
You can attach "rootfs_size_mb" to the request JSON. This isn't exposed in the Firmware Selector but please feel free to add a Pull Request for that feature.

hadifarnoud, 2025-02-18:
> Simplest way is to use the `--rootfs-size` option on owut, details at https://openwrt.org/docs/guide-user/installation/sysupgrade.owut#expanding_root_file_system
> 
> ```
> $ owut upgrade --rootfs-size 1024
> ```

it's limited to 1GB, right? I still think having the proper sized img would be easier and safer to do.

aparcar, 2025-02-18:
The server is limited to 1GB, other server instances may offer larger sizes

hadifarnoud, 2025-02-18:
@aparcar it worked! I modified the js code to include the rootfs_size_mb. 
```json
{
  "profile": "rpi-4",
  "target": "bcm27xx/bcm2711",
  "packages": [
    "base-files",
    "bcm27xx-gpu-fw",
    "bcm27xx-utils",
    "ca-bundle",
    "dropbear",
    "e2fsprogs",
    "firewall4",
    "fstools",
    "kmod-fs-vfat",
    "kmod-nft-offload",
    "kmod-nls-cp437",
    "kmod-nls-iso8859-1",
    "kmod-sound-arm-bcm2835",
    "kmod-sound-core",
    "kmod-usb-hid",
    "libc",
    "libgcc",
    "libustream-mbedtls",
    "logd",
    "mkf2fs",
    "mtd",
    "netifd",
    "nftables",
    "odhcp6c",
    "odhcpd-ipv6only",
    "opkg",
    "partx-utils",
    "ppp",
    "ppp-mod-pppoe",
    "procd-ujail",
    "uci",
    "uclient-fetch",
    "urandom-seed",
    "cypress-firmware-43455-sdio",
    "brcmfmac-nvram-43455-sdio",
    "kmod-brcmfmac",
    "wpad-basic-mbedtls",
    "kmod-usb-net-lan78xx",
    "kmod-r8169",
    "iwinfo",
    "luci",
    "luci-app-p910nd",
    "p910nd",
    "brcmfmac-firmware-usb",
    "kmod-tun",
    "kmod-usb-core",
    "kmod-usb-hid",
    "kmod-usb-net",
    "kmod-usb-net-cdc-ether",
    "kmod-usb-printer",
    "kmod-usb-storage",
    "kmod-usb-storage-uas",
    "kmod-usb-xhci-hcd",
    "kmod-usb3",
    "kmod-usb2",
    "kmod-usb-core",
    "kmod-usb-hid",
    "kmod-usb-net",
    "usbutils",
    "docker",
    "docker-compose",
    "dockerd",
    "dnsmasq-full",
    "curl",
    "ca-bundle",
    "ip-full",
    "ruby",
    "ruby-yaml",
    "kmod-tun",
    "kmod-inet-diag",
    "unzip",
    "kmod-nft-tproxy",
    "luci-compat",
    "luci",
    "luci-base"
  ],
  "defaults": "",
  "version_code": "r28427-6df0e3d02a",
  "version": "24.10.0",
  "rootfs_size_mb": "1024"
  "diff_packages": true,
  "client": "ofs/v5.0.0-2-g4336d760"
}
```

[image]

MarkBeharrell, 2025-02-23:
I also found I kept hitting the "Impossible package" so although I could get larger images built I had no idea what packages caused me issues so I wrote a simple script that does some cross checking and gives you a package list that should be cleaned and without the renamed or missing pacakges.

[OpenWRT-ASU-PackageChecker](https://github.com/MarkBeharrell/OpenWRT-ASU-PackageChecker/blob/trunk/packageChecker.sh)

BeBiMa, 2025-03-16:
@hadifarnoud , where did you add the entry?

hadifarnoud, 2025-03-16:
I used js in chrome console