openwrt/asu #1260: Unsupported target: ipq807x/generic - Xiaomi AX3600

Code-TT, 2025-02-18:
```
Error building the firmware image
Server response: Unsupported target: ipq807x/generic. The requested target was either dropped, is still being built or is not supported by the selected version. Please check the forums or try again later.

______________

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24106-10cc5fcd00",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "xiaomi,ax3600",
    "target": "ipq807x/generic",
    "version": "24.10.0",
    "packages": [
        "ath10k-firmware-qca9887-ct",
        "ath11k-firmware-ipq8074",
        "base-files",
        "batctl-tiny",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "firewall4",
        "fping",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ipq-wifi-xiaomi_ax3600",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct-smallbuffers",
        "kmod-ath11k",
        "kmod-ath11k-ahb",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-michael-mic",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha512",
        "kmod-fs-ext4",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc16",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-libphy",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-phy-aquantia",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-qca-nss-dp",
        "kmod-qca-ssdk",
        "kmod-qrtr",
        "kmod-qrtr-smd",
        "kmod-slhc",
        "kmod-thermal",
        "kmod-usb-core",
        "kmod-usb-dwc3",
        "kmod-usb-dwc3-qcom",
        "kmod-usb-xhci-hcd",
        "kmod-usb3",
        "libblkid",
        "libblobmsg-json",
        "libc",
        "libcomerr",
        "libext2fs",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "liblucihttp",
        "liblucihttp-ucode",
        "libmbedtls",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libopenssl",
        "libopenssl-conf",
        "libopenssl-legacy",
        "libsmartcols",
        "libss",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "libuuid",
        "logd",
        "losetup",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-usteer",
        "luci-base",
        "luci-light",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-batman-adv",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "nano-full",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-mbedtls",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "rpcd-mod-ucode",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-html",
        "ucode-mod-math",
        "ucode-mod-nl80211",
        "ucode-mod-rtnl",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "ucode-mod-uloop",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-openssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-25.037.68351-2ca3f19"
}

```

efahl, 2025-02-19:
```
The requested target was either dropped, is still being built or is not supported by the selected version. Please check the forums or try again later.
```
And what have you done, based upon the above message?

migueldeblas, 2025-02-24:
Will be this:

AX3600 Info:

OpenWrt 24.10 supports over 1970 devices. Support for over 100 new devices was added in addition to the device support by OpenWrt 23.05.

* Added support for OpenWrt One
Target changes
....
    **Renamed ipq807x target to qualcommax.**
....