openwrt/asu #1264: Error: Impossible package selection

deafmute420, 2025-02-20:
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "SNAPSHOT",
    "revision": "r28840-31dc43daf5",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "79b118b19bf7079c294332ff2fda35f3432f5254e0aeff972132ce4daf3afeee",
    "sha256_unsigned": "",
    "client": "luci/25.049.66340~7fb534c",
    "packages": {
        "adguardhome": "0.107.56-r1",
        "apk-mbedtls": "3.0.0_pre20250208-r1",
        "base-files": "1655~31dc43daf5",
        "bind-host": "9.20.5-r1",
        "block-mount": "2024.12.02~49d36ba2-r1",
        "busybox": "1.37.0-r4",
        "ca-bundle": "20240203-r1",
        "collectd-mod-cpu": "5.12.0-r53",
        "collectd-mod-interface": "5.12.0-r53",
        "collectd-mod-memory": "5.12.0-r53",
        "collectd-mod-ping": "5.12.0-r53",
        "collectd-mod-rrdtool": "5.12.0-r53",
        "collectd-mod-wireless": "5.12.0-r53",
        "curl": "8.12.1-r1",
        "ddns-scripts": "2.8.2-r63",
        "dnsmasq": "2.90-r4",
        "dockerd": "27.3.1-r1",
        "dropbear": "2024.86-r1",
        "e2fsprogs": "1.47.2-r1",
        "eip197-mini-firmware": "20241110-r1",
        "f2fsck": "1.16.0-r3",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "fitblk": "2",
        "fstools": "2024.12.02~49d36ba2-r1",
        "fwtool": "2019.11.12~8f7fe925-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "hostapd-common": "2025.02.09~c8c7d56a-r1",
        "https-dns-proxy": "2023.12.26-r5",
        "iwinfo": "2025.02.06~9cec6b4d-r1",
        "jansson4": "2.14-r3",
        "jshn": "2024.03.29~eb9bcb64-r1",
        "jsonfilter": "2024.01.23~594cfa86-r1",
        "kernel": "6.6.77~8e55049f1bff9e2037921fe3319b100e-r1",
        "kmod-crypto-acompress": "6.6.77-r1",
        "kmod-crypto-aead": "6.6.77-r1",
        "kmod-crypto-authenc": "6.6.77-r1",
        "kmod-crypto-crc32c": "6.6.77-r1",
        "kmod-crypto-des": "6.6.77-r1",
        "kmod-crypto-hash": "6.6.77-r1",
        "kmod-crypto-hmac": "6.6.77-r1",
        "kmod-crypto-hw-safexcel": "6.6.77-r1",
        "kmod-crypto-manager": "6.6.77-r1",
        "kmod-crypto-md5": "6.6.77-r1",
        "kmod-crypto-null": "6.6.77-r1",
        "kmod-crypto-sha1": "6.6.77-r1",
        "kmod-crypto-sha256": "6.6.77-r1",
        "kmod-crypto-sha512": "6.6.77-r1",
        "kmod-eeprom-at24": "6.6.77-r1",
        "kmod-fs-exfat": "6.6.77-r1",
        "kmod-fs-ntfs3": "6.6.77-r1",
        "kmod-gpio-button-hotplug": "6.6.77-r5",
        "kmod-hwmon-core": "6.6.77-r1",
        "kmod-hwmon-pwmfan": "6.6.77-r1",
        "kmod-i2c-mux-pca954x": "6.6.77-r1",
        "kmod-leds-gpio": "6.6.77-r1",
        "kmod-lib-crc-ccitt": "6.6.77-r1",
        "kmod-lib-crc32c": "6.6.77-r1",
        "kmod-lib-lzo": "6.6.77-r1",
        "kmod-libphy": "6.6.77-r1",
        "kmod-mt7996-233-firmware": "6.6.77.2025.02.14~e5fef138-r1",
[... 151 lines cut ...]
  runc (no such package):
    required by: containerd-1.7.22-r1[runc]
  libseccomp (no such package):
    required by: dockerd-27.3.1-r1[libseccomp]
  ip-full (no such package):
    required by: pbr-1.1.8-r10[ip-full]
  samba4-server (no such package):
    required by: luci-app-samba4-25.049.66340~7fb534c[samba4-server]
  collectd (no such package):
    required by: luci-app-statistics-25.049.66340~7fb534c[collectd]
  collectd-mod-iwinfo (no such package):
    required by: luci-app-statistics-25.049.66340~7fb534c[collectd-mod-iwinfo]
  collectd-mod-load (no such package):
    required by: luci-app-statistics-25.049.66340~7fb534c[collectd-mod-load]
  collectd-mod-network (no such package):
    required by: luci-app-statistics-25.049.66340~7fb534c[collectd-mod-network]
  rrdtool1 (no such package):
    required by: luci-app-statistics-25.049.66340~7fb534c[rrdtool1]
  ip (no such package):
    required by: wireguard-tools-1.0.20210914-r4[ip]
  libgnutls (no such package):
    required by: msmtp-1.8.26-r1[libgnutls]
make[2]: *** [Makefile:225: package_install] Error 43
make[1]: *** [Makefile:166: _call_manifest] Error 2
make: *** [Makefile:343: manifest] Error 2

deafmute420, 2025-02-20:
I've also been having problems with hostapd and my wifi radios not working anymore

andybjackson, 2025-02-27:
Check again. I had the same problem which has now resolved itself.

If at all possible, rather add to an existing issue than opening a new one. In this instance there were many issues opened against the same symptoms.

deafmute420, 2025-05-22:
> Check again. I had the same problem which has now resolved itself.
> 
> If at all possible, rather add to an existing issue than opening a new one. In this instance there were many issues opened against the same symptoms.

hey man mb, I kinda new here and just want to contribute.

closed 2025-05-22 by deafmute420