openwrt/asu #1270: Error: Impossible package selection, y'all updated base-files and kernel but still haven't fixed this, plz fix my wifi doesn't work anymore.

deafmute420, 2025-02-22:


Soromeister, 2025-02-22:
Similar issue with qualcommax/ipq807x:

Request:
```
POST /api/v1/build HTTP/1.1
Content-Length: 1553
Content-Type: application/json
Host: sysupgrade.openwrt.org
User-Agent: HTTPie

{
    "profile": "xiaomi,ax9000",
    "target": "qualcommax/ipq807x",
    "version": "SNAPSHOT",
    "version_code": "r28869-6936deac67",
    "defaults": "",
    "diff_packages": true,
    "filesystem": "squashfs",
    "packages": [
        "apk",
        "ath10k-firmware-qca9887-ct-full-htt",
        "ath11k-firmware-ipq8074",
        "ath11k-firmware-qcn9074",
        "base-files",
        "ca-bundle",
        "coreutils",
        "coreutils-uptime",
        "curl",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "firewall4",
        "fstools",
        "gre",
        "htop",
        "ipq-wifi-xiaomi_ax9000",
        "jq",
        "kmod-ath10k-ct",
        "kmod-ath11k-ahb",
        "kmod-ath11k-pci",
        "kmod-fs-ext4",
        "kmod-gpio-button-hotplug",
        "kmod-leds-gpio",
        "kmod-nft-offload",
        "kmod-phy-aquantia",
        "kmod-qca-nss-dp",
        "kmod-udptunnel6",
        "kmod-usb-dwc3",
        "kmod-usb-dwc3-qcom",
        "kmod-usb3",
        "libc",
        "libgcc",
        "libustream-wolfssl",
        "libwolfsslcpu-crypto5.7.6.e624513f",
        "logd",
        "losetup",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-irqbalance",
        "mtd",
        "ncat-full",
        "netifd",
        "nftables",
        "odhcp6c",
        "odhcpd-ipv6only",
        "ppp",
        "ppp-mod-pppoe",
        "uboot-envtools",
        "uci",
        "uclient-fetch",
[... 26 lines cut ...]
Request:
```
GET /api/v1/build/9c18842b198646f0ea1bcc7452379374704d9db6d28b31311e13bdddbf7209ef HTTP/1.1
Host: sysupgrade.openwrt.org
User-Agent: HTTPie
```
Response:
```
HTTP/1.1 500 Internal Server Error
Access-Control-Allow-Credentials: true
Access-Control-Allow-Headers: *
Access-Control-Allow-Methods: POST, GET, OPTIONS
Access-Control-Allow-Origin: *
Access-Control-Expose-Headers: Authorization
Connection: close
Content-Length: 3377
Content-Type: application/json
Date: Sat, 22 Feb 2025 11:59:06 GMT
Server: Caddy
Vary: Origin
X-Imagebuilder-Status: failed
X-Queue-Position: 0

{"detail":"Error: Impossible package selection","imagebuilder_status":"failed","request":{"distro":"openwrt","version":"SNAPSHOT","version_code":"r28869-6936deac67","target":"qualcommax/ipq807x","profile":"xiaomi_ax9000","packages":["apk","ath10k-firmware-qca9887-ct-full-htt","ath11k-firmware-ipq8074","ath11k-firmware-qcn9074","base-files","ca-bundle","coreutils","coreutils-uptime","curl","dnsmasq","dropbear","e2fsprogs","firewall4","fstools","gre","htop","ipq-wifi-xiaomi_ax9000","jq","kmod-ath10k-ct","kmod-ath11k-ahb","kmod-ath11k-pci","kmod-fs-ext4","kmod-gpio-button-hotplug","kmod-leds-gpio","kmod-nft-offload","kmod-phy-aquantia","kmod-qca-nss-dp","kmod-udptunnel6","kmod-usb-dwc3","kmod-usb-dwc3-qcom","kmod-usb3","libc","libgcc","libustream-wolfssl","libwolfsslcpu-crypto5.7.6.e624513f","logd","losetup","luci","luci-app-attendedsysupgrade","luci-app-irqbalance","mtd","ncat-full","netifd","nftables","odhcp6c","odhcpd-ipv6only","ppp","ppp-mod-pppoe","uboot-envtools","uci","uclient-fetch","urandom-seed","urngd","wget-ssl","wpad-wolfssl"],"packages_versions":{},"diff_packages":true,"defaults":"","rootfs_size_mb":null,"repositories":{},"repository_keys":[],"client":null},"stdout":"","stderr":"Generate local signing keys...\nWARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf\nWARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf\nread EC key\nwriting EC key\nWARNING: opening /builder/packages/packages.adb: No such file or directory\nPackage list missing or not up-to-date, generating it.\n\nBuilding package index...\nERROR: unable to select packages:\n  htop (no such package):\n    required by: world[htop]\n  libustream-wolfssl (no such package):\n    required by: world[libustream-wolfssl]\n  libwolfsslcpu-crypto5.7.6.e624513f (no such package):\n    required by: world[libwolfsslcpu-crypto5.7.6.e624513f]\n  nftables (no such package):\n    required by: world[nftables]\n  ppp (no such package):\n    required by: world[ppp]\n  ppp-mod-pppoe (no such package):\n    required by: world[ppp-mod-pppoe]\n  wpad-wolfssl (no such package):\n    required by: world[wpad-wolfssl]\n  nftables-json (no such package):\n    required by: firewall4-2024.12.18~18fc0ead-r1[nftables-json]\nmake[2]: *** [Makefile:226: package_install] Error 12\nmake[1]: *** [Makefile:167: _call_manifest] Error 2\nmake: *** [Makefile:344: manifest] Error 2\n","status":500,"error":"Traceback (most recent call last):\n  File \"/usr/local/lib/python3.12/site-packages/rq/worker.py\", line 1633, in perform_job\n    return_value = job.perform()\n                   ^^^^^^^^^^^^^\n  File \"/usr/local/lib/python3.12/site-packages/rq/job.py\", line 1331, in perform\n    self._result = self._execute()\n                   ^^^^^^^^^^^^^^^\n  File \"/usr/local/lib/python3.12/site-packages/rq/job.py\", line 1365, in _execute\n    result = self.func(*self.args, **self.kwargs)\n             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\n  File \"/app/asu/build.py\", line 245, in build\n    report_error(job, \"Impossible package selection\")\n  File \"/app/asu/util.py\", line 308, in report_error\n    raise RuntimeError(msg)\nRuntimeError: Impossible package selection\n","enqueued_at":"2025-02-22T11:56:29.927922","request_hash":"9c18842b198646f0ea1bcc7452379374704d9db6d28b31311e13bdddbf7209ef"}
```

MarkBeharrell, 2025-02-23:
I had the same and its due to the packages in the list being either not available or renamed etc. I built a script that can help:

[Discussion: OpenWrt Package Checker and AUS-Compatible List Generator](https://github.com/openwrt/asu/discussions/1271)

Soromeister, 2025-02-23:
> I had the same and its due to the packages in the list being either not available or renamed etc. I built a script that can help:
> 
> [Discussion: OpenWrt Package Checker and AUS-Compatible List Generator](https://github.com/openwrt/asu/discussions/1271)

Unfortunately, your tools uses the legacy `opkg` which hasn't been available since last year, ever since OpenWrt moved to `apk`.

deafmute420, 2025-02-24:
I got a Banana Pi-R4 and been using the snapshot for over 2 months and it's been working fine until about 2 weeks ago. I ended up rolling back to 24.10 and recompiled that with my package list but I've been using apk with the snapshot and it didn't compile with a package manager. 24.10 uses opkg so I remade it with that and it's been working like normal again.

rah66, 2025-02-25:
Similar problem.

Error: Impossible package selection
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "SNAPSHOT",
    "revision": "r28862-edfe589811",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "9b162e1b564f2922ebc49e4a923243b9df94a219bd802b0996d0d78bbf966f43",
    "sha256_unsigned": "",
    "client": "luci/25.049.66340~7fb534c",
    "packages": {
        "apk-mbedtls": "3.0.0_pre20250208-r1",
        "base-files": "1655~edfe589811",
        "ca-bundle": "20240203-r1",
        "dnsmasq": "2.90-r4",
        "dropbear": "2024.86-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "fitblk": "2",
        "fstools": "2024.12.02~49d36ba2-r1",
        "kernel": "6.6.77~8e55049f1bff9e2037921fe3319b100e-r1",
        "kmod-crypto-hw-safexcel": "6.6.77-r1",
        "kmod-gpio-button-hotplug": "6.6.77-r5",
        "kmod-leds-gpio": "6.6.77-r1",
        "kmod-mt7915e": "6.6.77.2025.02.14~e5fef138-r1",
        "kmod-mt7981-firmware": "6.6.77.2025.02.14~e5fef138-r1",
        "kmod-nft-offload": "6.6.77-r1",
        "kmod-phy-aquantia": "6.6.77-r1",
        "libc": "1.2.5-r4",
        "logd": "2024.04.26~85f10530-r1",
        "luci": "25.049.66340~7fb534c",
        "luci-app-attendedsysupgrade": "25.049.66340~7fb534c",
        "luci-app-commands": "25.049.66340~7fb534c",
        "luci-app-firewall": "25.049.66340~7fb534c",
        "luci-app-package-manager": "25.049.66340~7fb534c",
        "luci-app-ttyd": "25.049.66340~7fb534c",
        "luci-base": "25.049.66340~7fb534c",
        "luci-i18n-attendedsysupgrade-ru": "25.049.66340~7fb534c",
        "luci-i18n-base-ru": "25.049.66340~7fb534c",
        "luci-i18n-commands-ru": "25.049.66340~7fb534c",
        "luci-i18n-ddns-ru": "25.049.66340~7fb534c",
        "luci-i18n-filemanager-ru": "25.049.66340~7fb534c",
        "luci-i18n-firewall-ru": "25.049.66340~7fb534c",
        "luci-i18n-package-manager-ru": "25.049.66340~7fb534c",
        "luci-i18n-ttyd-ru": "25.049.66340~7fb534c",
        "luci-light": "25.049.66340~7fb534c",
        "luci-mod-admin-full": "25.049.66340~7fb534c",
        "luci-mod-network": "25.049.66340~7fb534c",
        "luci-mod-status": "25.049.66340~7fb534c",
        "luci-mod-system": "25.049.66340~7fb534c",
        "luci-proto-ipv6": "25.049.66340~7fb534c",
        "luci-proto-ppp": "25.049.66340~7fb534c",
        "luci-proto-relay": "25.049.66340~7fb534c",
        "luci-theme-bootstrap": "25.049.66340~7fb534c",
        "mt7981-wo-firmware": "20241110-r1",
        "mtd": "26",
        "netifd": "2024.12.17~ea01ed41-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "ppp": "2.5.2-r1",
        "ppp-mod-pppoe": "2.5.2-r1",
        "procd-ujail": "2025.01.30~7fcb5a27-r1",
        "ttyd": "1.7.3-r1",
        "uboot-envtools": "2025.01-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "uclient-fetch": "2024.10.22~88ae8f20-r1",
        "urandom-seed": "3",
        "urngd": "2023.11.01~44365eb1-r1",
        "wpad-basic-mbedtls": "2025.02.09~c8c7d56a-r1"
    },
    "profile": "xiaomi,mi-router-ax3000t",
    "target": "mediatek/filogic",
    "version": "SNAPSHOT",
    "diff_packages": true,
    "filesystem": "squashfs"
}
STDERR:
Generate local signing keys...
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
read EC key
writing EC key
WARNING: opening /builder/packages/packages.adb: No such file or directory
Package list missing or not up-to-date, generating it.

Building package index...
ERROR: unable to select packages:
  ppp (no such package):
    required by: world[ppp]
  ppp-mod-pppoe (no such package):
    required by: world[ppp-mod-pppoe]
  nftables-json (no such package):
    required by: firewall4-2024.12.18~18fc0ead-r1[nftables-json]
make[2]: *** [Makefile:226: package_install] Error 8
make[1]: *** [Makefile:167: _call_manifest] Error 2
make: *** [Makefile:344: manifest] Error 2

rah66, 2025-03-01:
> Check again. I had the same problem which has now resolved itself.
> 
> If at all possible, rather add to an existing issue than opening a new one. In this instance there were many issues opened against the same symptoms.

The problem is not resolved for device - [Xiaomi AX3000T](https://firmware-selector.openwrt.org/?version=SNAPSHOT&target=mediatek%2Ffilogic&id=xiaomi_mi-router-ax3000t). 

This can be easily verified by deleting, for example, **luci** package and running the build.

Soromeister, 2025-03-02:
Problem is not resolved also for AX9000 device. You can test this in the same way, remove `luci` package and see that it fails with the same error message.

rah66, 2025-03-02:
> Problem is not resolved also for AX9000 device. You can test this in the same way, remove `luci` package and see that it fails with the same error message.

AX3000T & AX9000 corrected

Soromeister, 2025-03-03:
Confirming fix for AX9000. Thank you @rah66.

closed 2025-05-22 by deafmute420