openwrt/asu #1298: sysupgrade failed 23.05 to 24.10

petersie, 2025-03-27:
Failed Attended Sysupgrade on APU2.

Currently running: 23.05.5 - r24106-10cc5fcd00

---

[Bearbeiten6.txt](https://github.com/user-attachments/files/19489035/Bearbeiten6.txt)

thank you
sigi

Patt92, 2025-03-28:
the log obviously says at the end, remove it from the modules in the list below the build packages
" * opkg_install_cmd: Cannot install package perlbase-arybase."

petersie, 2025-04-01:
Thank you, that helps.

But now the following error occurs:

.......
Signature check passed.
Pseudo file "dev" exists in source filesystem "/builder/build_dir/target-x86_64_musl/root-x86/dev".
Ignoring, exclude it (-e/-ef) to override.
error: ext4_allocate_best_fit_partial: failed to allocate 2 blocks, out of space?
make[3]: *** [/builder/include/image.mk:350: /builder/build_dir/target-x86_64_musl/linux-x86_64/root.ext4] Error 1
make[2]: *** [Makefile:208: build_image] Error 2
make[1]: *** [Makefile:146: _call_image] Error 2
make: *** [Makefile:262: image] Error 2
........

[full_output.txt](https://github.com/user-attachments/files/19552579/full_output.txt)

And no, there is enought space on disk and memory.

sigi

BeBiMa, 2025-04-02:
Similiar problem here ( since weeks!! ).

[ASU.txt](https://github.com/user-attachments/files/19567546/ASU.txt)

ayubwaheed, 2025-04-15:
upgrade fail from 23.05 to 24.10  through web systupgrade uitily

[23.05.5 - r24106-10cc5fcd00_upgrdae.txt](https://github.com/user-attachments/files/19752197/23.05.5.-.r24106-10cc5fcd00_upgrdae.txt)