openwrt/asu #1302: firmware-selector Error: Could not set up ImageBuilder

Soromeister, 2025-04-04:
Cannot use https://firmware-selector.openwrt.org to build images. I am trying to generate the below image:
```
Model: Xiaomi AX9000
Platform: qualcommax/ipq807x
Version: SNAPSHOT (r29139-65b8a978de)
Installed Packages:
apk ath10k-firmware-qca9887-ct-full-htt ath11k-firmware-ipq8074 ath11k-firmware-qcn9074 base-files ca-bundle coreutils curl dnsmasq dropbear e2fsprogs firewall4 fstools gre htop ipq-wifi-xiaomi_ax9000 jq kmod-ath10k-ct kmod-ath11k-ahb kmod-ath11k-pci kmod-fs-ext4 kmod-gpio-button-hotplug kmod-leds-gpio kmod-nft-offload kmod-phy-aquantia kmod-qca-nss-dp kmod-udptunnel6 kmod-usb-dwc3 kmod-usb-dwc3-qcom kmod-usb3 libc libgcc libustream-wolfssl libwolfsslcpu-crypto logd losetup luci luci-app-attendedsysupgrade luci-app-irqbalance mtd ncat-full netifd nftables odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe uboot-envtools uci uclient-fetch urandom-seed urngd wget-ssl wpad-wolfssl
```

Output of STDERR:
```
STDERR
--2025-04-04 22:59:34--  http://downloads.openwrt.org/snapshots/targets/qualcommax/ipq807x/sha256sums
Connecting to 127.0.0.1:3128... connected.
Proxy request sent, awaiting response... 200 OK
Length: 5012240 (4.8M) [text/plain]
Saving to: ‘sha256sums’

     0K .......... .......... .......... .......... ..........  1%  386M 0s
    50K .......... .......... .......... .......... ..........  2%  344M 0s
   100K .......... .......... .......... .......... ..........  3%  442M 0s
   150K .......... .......... .......... .......... ..........  4%  468M 0s
   200K .......... .......... .......... .......... ..........  5%  382M 0s
   250K .......... .......... .......... .......... ..........  6%  494M 0s
   300K .......... .......... .......... .......... ..........  7%  463M 0s
   350K .......... .......... .......... .......... ..........  8%  354M 0s
   400K .......... .......... .......... .......... ..........  9%  254M 0s
   450K .......... .......... .......... .......... .......... 10%  340M 0s
   500K .......... .......... .......... .......... .......... 11%  247M 0s
   550K .......... .......... .......... .......... .......... 12%  259M 0s
   600K .......... .......... .......... .......... .......... 13%  253M 0s
   650K .......... .......... .......... .......... .......... 14%  339M 0s
   700K .......... .......... .......... .......... .......... 15%  327M 0s
   750K .......... .......... .......... .......... .......... 16%  198M 0s
   800K .......... .......... .......... .......... .......... 17%  254M 0s
   850K .......... .......... .......... .......... .......... 18%  275M 0s
   900K .......... .......... .......... .......... .......... 19%  292M 0s
   950K .......... .......... .......... .......... .......... 20%  242M 0s
  1000K .......... .......... .......... .......... .......... 21%  279M 0s
  1050K .......... .......... .......... .......... .......... 22%  283M 0s
  1100K .......... .......... .......... .......... .......... 23%  258M 0s
  1150K .......... .......... .......... .......... .......... 24%  237M 0s
  1200K .......... .......... .......... .......... .......... 25%  347M 0s
  1250K .......... .......... .......... .......... .......... 26%  240M 0s
  1300K .......... .......... .......... .......... .......... 27%  274M 0s
  1350K .......... .......... .......... .......... .......... 28%  278M 0s
  1400K .......... .......... .......... .......... .......... 29%  245M 0s
  1450K .......... .......... .......... .......... .......... 30%  288M 0s
  1500K .......... .......... .......... .......... .......... 31%  234M 0s
  1550K .......... .......... .......... .......... .......... 32%  246M 0s
  1600K .......... .......... .......... .......... .......... 33%  299M 0s
  1650K .......... .......... .......... .......... .......... 34%  226M 0s
  1700K .......... .......... .......... .......... .......... 35%  329M 0s
  1750K .......... .......... .......... .......... .......... 36%  224M 0s
  1800K .......... .......... .......... .......... .......... 37%  229M 0s
  1850K .......... .......... .......... .......... .......... 38%  367M 0s
  1900K .......... .......... .......... .......... .......... 39%  272M 0s
  1950K .......... .......... .......... .......... .......... 40%  279M 0s
  2000K .......... .......... .......... .......... .......... 41%  292M 0s
  2050K .......... .......... .......... .......... .......... 42%  266M 0s
  2100K .......... .......... .......... .......... .......... 43%  241M 0s
  2150K .......... .......... .......... .......... .......... 44%  302M 0s
  2200K .......... .......... .......... .......... .......... 45%  251M 0s
  2250K .......... .......... .......... .......... .......... 46%  341M 0s
  2300K .......... .......... .......... .......... .......... 48%  224M 0s
  2350K .......... .......... .......... .......... .......... 49%  241M 0s
  2400K .......... .......... .......... .......... .......... 50%  325M 0s
  2450K .......... .......... .......... .......... .......... 51%  243M 0s
  2500K .......... .......... .......... .......... .......... 52%  327M 0s
  2550K .......... .......... .......... .......... .......... 53%  217M 0s
[... 55 lines cut ...]
     0K                                                       100%  116M=0s

2025-04-04 22:59:34 (116 MB/s) - ‘sha256sums.asc’ saved [299/299]

gpg: directory '/builder/.gnupg' created
gpg: keybox '/builder/.gnupg/pubring.kbx' created
gpg: key 1D53D1877742E911: 3 signatures not checked due to missing keys
gpg: /builder/.gnupg/trustdb.gpg: trustdb created
gpg: key 1D53D1877742E911: public key "OpenWrt Build System (Nitrokey3) <contact@openwrt.org>" imported
gpg: key 28A39BC32074BE7A: 2 signatures not checked due to missing keys
gpg: key 28A39BC32074BE7A: public key "OpenWrt Build System (PGP key for 19.07 release builds) <pgpsign-19.07@openwrt.org>" imported
gpg: key CD84BCED626471F1: 2 signatures not checked due to missing keys
gpg: key CD84BCED626471F1: public key "OpenWrt Build System (PGP key for unattended snapshot builds) <pgpsign-snapshots@openwrt.org>" imported
gpg: key 88CA59E88F681580: public key "OpenWrt Build System (PGP key for 21.02 release builds) <pgpsign-21.02@openwrt.org>" imported
gpg: key CD54E82DADB3684D: public key "OpenWrt Build System (GnuPGP key for 22.03 release builds) <pgpsign-22.03@openwrt.org>" imported
gpg: Total number processed: 5
gpg:               imported: 5
gpg: no ultimately trusted keys found
rm: cannot remove '/builder/keys/CD54E82DADB3684D.asc': Permission denied
rm: cannot remove '/builder/keys/0x1D53D1877742E911.asc': Permission denied
rm: cannot remove '/builder/keys/2074BE7A.asc': Permission denied
rm: cannot remove '/builder/keys/88CA59E8.asc': Permission denied
rm: cannot remove '/builder/keys/626471F1.asc': Permission denied
STDOUT is empty
```

efahl, 2025-04-05:
Fixed in https://github.com/openwrt/docker/commit/648a9bd5ba25e02c53995e95401b77734c46a1eb

Soromeister, 2025-04-06:
Confirmed fix.

closed 2025-04-06 by Soromeister