openwrt/asu #1308: ASU 23.05.5 -> 24.10.1 failed with an error message: Unsupported version: 24.10.1

StentMan, 2025-04-14:
The upgrade to 24.10.1 is the only firmware version that appears in the drop-down list.

{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "23.05",
    "revision": "r24106-10cc5fcd00",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/git-25.098.82158-b32b5fc",
    "packages": {
        "iwinfo": "2023-07-01-ca79f641-1",
        "luci-app-statistics": "git-25.098.82158-b32b5fc",
        "liblucihttp": "2023-03-15-9b5b683f-1",
        "cgi-io": "2022-08-10-901b0f04-21",
        "luci-lib-base": "git-25.098.82158-b32b5fc",
        "libc": "1.2.4-4",
        "opkg": "2022-02-24-d038e5b6-2",
        "wpad-basic-mbedtls": "2023-09-08-e5ccbfc6-8",
        "luci-app-opkg": "git-25.098.82158-b32b5fc",
        "ubus": "2023-06-05-f787c97b-1",
        "libuci": "2023-08-10-5781664d-1",
        "rpcd": "2023-07-01-c07ab2f9-1",
        "busybox": "1.36.1-1",
        "libiwinfo": "2023-07-01-ca79f641-1",
        "luci-lib-ip": "git-25.098.82158-b32b5fc",
        "kmod-nft-fib": "5.15.167-1",
        "kmod-nfnetlink": "5.15.167-1",
        "kmod-crypto-hash": "5.15.167-1",
        "kmod-nf-reject6": "5.15.167-1",
        "luci-app-https-dns-proxy": "2023.12.26",
        "luci-mod-system": "git-25.098.82158-b32b5fc",
        "kmod-nf-flow": "5.15.167-1",
        "libnl-tiny": "2023-07-27-bc92a280-1",
        "kmod-lib-crc-ccitt": "5.15.167-1",
        "px5g-mbedtls": "10",
        "getrandom": "2022-08-13-4c7b720b-2",
        "ucode-mod-ubus": "2024-07-11-1a8a0bcf-3",
        "luci-theme-bootstrap": "git-25.098.82158-b32b5fc",
        "kmod-pppoe": "5.15.167-1",
        "kmod-pppox": "5.15.167-1",
        "kmod-nf-reject": "5.15.167-1",
        "procd-ujail": "2023-06-25-2db83655-2",
        "base-files": "1562-r24106-10cc5fcd00",
        "kmod-nf-nat": "5.15.167-1",
        "kmod-crypto-crc32c": "5.15.167-1",
        "ucode-mod-uci": "2024-07-11-1a8a0bcf-3",
        "libustream-mbedtls": "2023-02-25-498f6e26-1",
        "netifd": "2024-01-04-c18cc79d-2",
        "firewall4": "2023-09-01-598d9fbb-1",
        "uboot-envtools": "2023.04-1",
        "dnsmasq": "2.90-2",
        "procd": "2023-06-25-2db83655-2",
        "ubusd": "2023-06-05-f787c97b-1",
        "kmod-crypto-acompress": "5.15.167-1",
        "rpcd-mod-ucode": "2023-07-01-c07ab2f9-1",
        "ucode-mod-math": "2024-07-11-1a8a0bcf-3",
        "kmod-lib-crc32c": "5.15.167-1",
        "luci-mod-status": "git-25.098.82158-b32b5fc",
        "libmbedtls": "2.28.9-1",
        "kmod-nft-nat": "5.15.167-1",
        "https-dns-proxy": "2023.12.26-1",
        "luci-app-firewall": "git-25.098.82158-b32b5fc",
        "libubus": "2023-06-05-f787c97b-1",
        "libmnl": "1.0.5-1",
        "jansson": "2.14-3",
        "mwlwifi-firmware-88w8864": "2023-11-20-2a5a4ae3-1",
        "ubi-utils": "2.1.5-1",
        "odhcp6c": "2023-05-12-bcd28363-20",
        "fstools": "2023-02-28-bfe882d5-1",
        "uclient-fetch": "2023-04-13-007d9454-1",
        "uci": "2023-08-10-5781664d-1",
        "ucode-mod-fs": "2024-07-11-1a8a0bcf-3",
        "luci-ssl": "git-25.098.82158-b32b5fc",
        "dropbear": "2022.82-6",
        "libnftnl": "1.2.6-1",
        "rpcd-mod-file": "2023-07-01-c07ab2f9-1",
        "mtd": "26",
        "odhcpd-ipv6only": "2023-10-24-d8118f6e-1",
        "procd-seccomp": "2023-06-25-2db83655-2",
        "libjson-script": "2023-05-23-75a3b870-1",
        "libiwinfo-data": "2023-07-01-ca79f641-1",
        "rpcd-mod-luci": "20240305-1",
        "ucode": "2024-07-11-1a8a0bcf-3",
        "kmod-nf-log": "5.15.167-1",
        "urandom-seed": "3",
        "luci-proto-ppp": "git-25.098.82158-b32b5fc",
        "luci-mod-admin-full": "git-25.098.82158-b32b5fc",
        "ppp": "2.4.9.git-2021-01-04-4",
        "luci-base": "git-25.098.82158-b32b5fc",
        "kmod-gpio-button-hotplug": "5.15.167-3",
        "logd": "2022-08-13-4c7b720b-2",
        "kmod-nf-log6": "5.15.167-1",
        "luci-proto-ipv6": "git-25.098.82158-b32b5fc",
        "openwrt-keyring": "2022-03-25-62471e69-2",
        "luci-app-attendedsysupgrade": "git-25.098.82158-b32b5fc",
        "jshn": "2023-05-23-75a3b870-1",
        "nftables-json": "1.0.8-1",
        "kmod-mwlwifi": "5.15.167+2023-11-20-2a5a4ae3-1",
        "wifischedule": "1-3",
        "luci-app-wifischedule": "git-25.098.82158-b32b5fc",
        "kmod-nft-offload": "5.15.167-1",
        "kmod-ppp": "5.15.167-1",
        "libjson-c": "0.16-3",
        "uhttpd": "2023-06-25-34a8a74d-2",
        "kmod-nf-conntrack": "5.15.167-1",
        "usign": "2020-05-23-f1f65026-1",
        "libblobmsg-json": "2023-05-23-75a3b870-1",
        "luci-lib-nixio": "git-25.098.82158-b32b5fc",
        "ca-bundle": "20230311-1",
        "libuclient": "2023-04-13-007d9454-1",
        "luci-lib-jsonc": "git-25.098.82158-b32b5fc",
        "ucode-mod-html": "1",
        "luci": "git-25.098.82158-b32b5fc",
        "kmod-nf-conntrack6": "5.15.167-1",
        "luci-light": "git-25.098.82158-b32b5fc",
        "kmod-lib-lzo": "5.15.167-1",
        "ubox": "2022-08-13-4c7b720b-2",
        "kernel": "5.15.167-1-fee90719398eed2badb627b27170f906",
        "rpcd-mod-iwinfo": "2023-07-01-c07ab2f9-1",
        "luci-mod-network": "git-25.098.82158-b32b5fc",
        "kmod-nft-core": "5.15.167-1",
        "libucode": "2024-07-11-1a8a0bcf-3",
        "uhttpd-mod-ubus": "2023-06-25-34a8a74d-2",
        "fwtool": "2019-11-12-8f7fe925-1",
        "jsonfilter": "2024-01-23-594cfa86-1",
        "liblucihttp-ucode": "2023-03-15-9b5b683f-1",
        "libubox": "2023-05-23-75a3b870-1",
        "wireless-regdb": "2024.10.07-1",
        "urngd": "2023-11-01-44365eb1-1",
        "kmod-slhc": "5.15.167-1",
        "rpcd-mod-rrdns": "20170710",
        "ppp-mod-pppoe": "2.4.9.git-2021-01-04-4"
    },
    "profile": "linksys,wrt1200ac",
    "target": "mvebu/cortexa9",
    "version": "23.05.5",
    "diff_packages": true,
    "filesystem": "squashfs"
}

StentMan, 2025-04-14:
I should've added that all packages on the current versions are up to date. I actually upgraded from 23.04.xx to 23.05.5 just before attempting the upgrade to 24.10.

nasm27, 2025-04-14:
For Asus Router TUF4200 if we upgrade to version 24.10.0, we will "kill" the router (known issue). The only way is move from 23.05.5 to 24.10.1, but I got the same  error as described on the top.
Any solution using ASU?

AreYouLoco, 2025-04-15:
See https://github.com/openwrt/asu/issues/1327

nasm27, 2025-04-15:
Apparently the issue is more or less identified, but solution is not easy
to apply. At least I don't know how.
Thanks

Best Regards
Nelson Marto
963858754

AreYouLoco, 2025-04-15:
Solution must be applied on server side so there is nothing you can do for now

StentMan, 2025-04-17:
Seems like I was able to upgrade today, only to run into some other issue (couldn't connect to WAN, I'm using PPPoE) which resulted in me being forced to downgrade back to 23.05.5, so I guess this issue is fixed?

closed 2025-04-17 by StentMan