openwrt/asu #1314: Attended Sysupgrade: Error building the firmware image Server response: Unsupported version: 24.10.1

mrleongalaxyum, 2025-04-15:
Xiaomi R4A Gigabit:

```
Powered by [LuCI openwrt-23.05 branch (git-24.264.56413-c7a3562)](https://github.com/openwrt/luci) / [OpenWrt 23.05.5 (r24106-10cc5fcd00)](https://openwrt.org/)
Error building the firmware image
Server response: Unsupported version: 24.10.1

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "23.05",
    "revision": "r24106-10cc5fcd00",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/git-25.098.82158-b32b5fc",
    "packages": {
        "luci-proto-wireguard": "git-25.053.28810-8216ff5",
        "iwinfo": "2023-07-01-ca79f641-1",
        "liblucihttp": "2023-03-15-9b5b683f-1",
        "cgi-io": "2022-08-10-901b0f04-21",
        "opkg": "2022-02-24-d038e5b6-2",
        "luci-app-opkg": "git-24.148.43905-2891ca4",
        "ubus": "2023-06-05-f787c97b-1",
        "libuci": "2023-08-10-5781664d-1",
        "rpcd": "2023-07-01-c07ab2f9-1",
        "busybox": "1.36.1-1",
        "libiwinfo": "2023-07-01-ca79f641-1",
        "kmod-nft-fib": "5.15.167-1",
        "kmod-nfnetlink": "5.15.167-1",
        "kmod-crypto-hash": "5.15.167-1",
        "kmod-nf-reject6": "5.15.167-1",
        "luci-mod-system": "git-24.067.01860-7a82b2f",
        "kmod-nf-flow": "5.15.167-1",
        "libnl-tiny": "2023-07-27-bc92a280-1",
        "kmod-lib-crc-ccitt": "5.15.167-1",
        "px5g-mbedtls": "10",
        "getrandom": "2022-08-13-4c7b720b-2",
        "ucode-mod-ubus": "2024-07-11-1a8a0bcf-3",
        "luci-theme-bootstrap": "git-24.086.46634-1ffe078",
        "kmod-pppoe": "5.15.167-1",
        "kmod-pppox": "5.15.167-1",
        "kmod-nf-reject": "5.15.167-1",
        "procd-ujail": "2023-06-25-2db83655-2",
        "base-files": "1562-r24106-10cc5fcd00",
        "kmod-nf-nat": "5.15.167-1",
        "kmod-crypto-crc32c": "5.15.167-1",
        "ucode-mod-uci": "2024-07-11-1a8a0bcf-3",
        "libustream-mbedtls": "2023-02-25-498f6e26-1",
        "netifd": "2024-01-04-c18cc79d-2",
        "firewall4": "2023-09-01-598d9fbb-1",
        "dnsmasq": "2.90-2",
        "procd": "2023-06-25-2db83655-2",
        "ubusd": "2023-06-05-f787c97b-1",
        "kmod-crypto-acompress": "5.15.167-1",
        "rpcd-mod-ucode": "2023-07-01-c07ab2f9-1",
        "ucode-mod-math": "2024-07-11-1a8a0bcf-3",
        "kmod-lib-crc32c": "5.15.167-1",
        "luci-mod-status": "git-24.212.61237-b6da3f2",
        "libmbedtls": "2.28.9-1",
        "kmod-nft-nat": "5.15.167-1",
        "luci-app-firewall": "git-24.067.01746-69867db",
        "libubus": "2023-06-05-f787c97b-1",
        "libmnl": "1.0.5-1",
        "jansson": "2.14-3",
        "ubi-utils": "2.1.5-1",
        "odhcp6c": "2023-05-12-bcd28363-20",
        "fstools": "2023-02-28-bfe882d5-1",
        "uclient-fetch": "2023-04-13-007d9454-1",
        "uci": "2023-08-10-5781664d-1",
        "ucode-mod-fs": "2024-07-11-1a8a0bcf-3",
        "luci-ssl": "git-23.035.26083-7550ad6",
        "dropbear": "2022.82-6",
        "libnftnl": "1.2.6-1",
        "rpcd-mod-file": "2023-07-01-c07ab2f9-1",
        "mtd": "26",
        "odhcpd-ipv6only": "2023-10-24-d8118f6e-1",
        "procd-seccomp": "2023-06-25-2db83655-2",
        "libjson-script": "2023-05-23-75a3b870-1",
        "libiwinfo-data": "2023-07-01-ca79f641-1",
        "ucode": "2024-07-11-1a8a0bcf-3",
        "rpcd-mod-luci": "20240305-1",
        "kmod-nf-log": "5.15.167-1",
        "urandom-seed": "3",
        "luci-proto-ppp": "git-24.135.44542-f1ec9c2",
        "luci-mod-admin-full": "git-19.253.48496-3f93650",
        "ppp": "2.4.9.git-2021-01-04-4",
        "luci-base": "git-24.264.56413-c7a3562",
        "kmod-leds-gpio": "5.15.167-1",
        "kmod-gpio-button-hotplug": "5.15.167-3",
        "logd": "2022-08-13-4c7b720b-2",
        "kmod-mt7603": "5.15.167+2024-04-03-1e336a85-1",
        "kmod-nf-log6": "5.15.167-1",
        "luci-proto-ipv6": "git-24.086.45108-51aee90",
        "openwrt-keyring": "2022-03-25-62471e69-2",
        "luci-app-attendedsysupgrade": "git-25.098.82158-b32b5fc",
        "irqbalance": "1.9.4-1",
        "jshn": "2023-05-23-75a3b870-1",
        "nftables-json": "1.0.8-1",
        "kmod-ppp": "5.15.167-1",
        "kmod-nft-offload": "5.15.167-1",
        "wpad-mbedtls": "2023-09-08-e5ccbfc6-8",
        "libjson-c": "0.16-3",
        "uhttpd": "2023-06-25-34a8a74d-2",
        "kmod-nf-conntrack": "5.15.167-1",
        "usign": "2020-05-23-f1f65026-1",
        "libblobmsg-json": "2023-05-23-75a3b870-1",
        "ca-bundle": "20230311-1",
        "libuclient": "2023-04-13-007d9454-1",
        "ucode-mod-html": "1",
        "luci": "git-23.051.66410-a505bb1",
        "kmod-nf-conntrack6": "5.15.167-1",
        "luci-light": "git-23.024.33244-34dee82",
        "kmod-lib-lzo": "5.15.167-1",
        "ubox": "2022-08-13-4c7b720b-2",
        "kernel": "5.15.167-1-144de9e5c1a8813b724b14faa054d9f0",
        "rpcd-mod-iwinfo": "2023-07-01-c07ab2f9-1",
        "kmod-mt76x2": "5.15.167+2024-04-03-1e336a85-1",
        "luci-mod-network": "git-24.264.56960-63ba3cb",
        "kmod-nft-core": "5.15.167-1",
        "libucode": "2024-07-11-1a8a0bcf-3",
        "uhttpd-mod-ubus": "2023-06-25-34a8a74d-2",
        "fwtool": "2019-11-12-8f7fe925-1",
        "jsonfilter": "2024-01-23-594cfa86-1",
        "liblucihttp-ucode": "2023-03-15-9b5b683f-1",
        "libubox": "2023-05-23-75a3b870-1",
        "hostapd-common": "2023-09-08-e5ccbfc6-8",
        "urngd": "2023-11-01-44365eb1-1",
        "kmod-slhc": "5.15.167-1",
        "rpcd-mod-rrdns": "20170710",
        "ppp-mod-pppoe": "2.4.9.git-2021-01-04-4"
    },
    "profile": "xiaomi,mi-router-4a-gigabit",
    "target": "ramips/mt7621",
    "version": "23.05.5",
    "diff_packages": true,
    "filesystem": "squashfs"
}
```

dwhitemv25, 2025-04-15:
Looks like someone forgot to tell [Sysupgrade](https://sysupgrade.openwrt.org/) the release announcement was going out... 24.10.1 isn't listed on its status page yet.

crza, 2025-04-16:
as far as I read the source code, the application just needs a restart to reload versions.json file from downloads.openwrt.org

@aparcar are you responsible for this?

dwhitemv25, 2025-04-18:
As of right now (4/18/2025 @ 19:32 UTC), 24.10.1 now listed on sysupgrade's status page, but attempting to run attended sysupgrade results in a 500 error trying to fetch `https://sysupgrade.openwrt.org/json/v1/overview.json`.

dwhitemv25, 2025-04-23:
I was able to run sysupgrade yesterday, so this ticket can be closed.

closed 2025-04-24 by mrleongalaxyum