openwrt/asu #1316: Error building the firmware image Server response: Unsupported version: 24.10.1- qemu-standard-pc-q35-ich9-2009-X86/64

madswaord, 2025-04-15:
Error building the firmware image
Server response: Unsupported version: 24.10.1

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.103.51521~2ac26e5",
    "packages": {
        "luci-proto-wireguard": "25.103.51521~2ac26e5",
        "luci-theme-material": "25.103.51521~2ac26e5",
        "libopenssl": "3.0.16-r1",
        "liblucihttp": "2023.03.15~9b5b683f-r1",
        "cgi-io": "2022.08.10~901b0f04-r21",
        "libc": "1.2.5-r4",
        "luci-app-nlbwmon": "25.103.51521~2ac26e5",
        "mkf2fs": "1.16.0-r3",
        "opkg": "2024.10.16~38eccbb1-r1",
        "kmod-ixgbe": "6.6.73-r1",
        "libiwinfo": "2024.10.20~b94f066e-r1",
        "luci-mod-system": "25.103.51521~2ac26e5",
        "libnl-tiny": "2025.03.19~c0df580a-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "luci-theme-openwrt": "25.103.51521~2ac26e5",
        "blockdev": "2.40.2-r1",
        "luci-theme-bootstrap": "25.103.51521~2ac26e5",
        "kmod-pppoe": "6.6.73-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "partx-utils": "2.40.2-r1",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "libf2fs": "1.16.0-r3",
        "kmod-r8169": "6.6.73-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "dnsmasq": "2.90-r4",
        "kmod-crypto-acompress": "6.6.73-r1",
        "luci-mod-status": "25.103.51521~2ac26e5",
        "luci-i18n-sqm-zh-cn": "25.103.51521~2ac26e5",
        "block-mount": "2024.07.14~408c2cc4-r1",
        "kmod-fs-vfat": "6.6.73-r1",
        "luci-app-firewall": "25.103.51521~2ac26e5",
        "grub2-efi": "2.12-r1",
        "libmbedtls": "3.6.3-r1",
        "grub2-bios-setup": "2.12-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "grub2": "2.12-r1",
        "losetup": "2.40.2-r1",
        "luci-ssl": "25.103.51521~2ac26e5",
        "dropbear": "2024.86-r1",
        "luci-theme-openwrt-2020": "25.103.51521~2ac26e5",
        "adguardhome": "0.107.57-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "luci-app-sqm": "25.103.51521~2ac26e5",
        "luci-app-package-manager": "25.103.51521~2ac26e5",
        "parted": "3.6-r1",
        "urandom-seed": "3",
        "luci-i18n-package-manager-zh-cn": "25.103.51521~2ac26e5",
        "luci-proto-ppp": "25.103.51521~2ac26e5",
        "ppp": "2.5.1-r1",
        "luci-mod-admin-full": "25.103.51521~2ac26e5",
        "cfdisk": "2.40.2-r1",
        "luci-base": "25.103.51521~2ac26e5",
        "logd": "2024.04.26~85f10530-r1",
        "lucky": "2.15.7-1",
        "kmod-tg3": "6.6.73-r1",
        "luci-proto-ipv6": "25.103.51521~2ac26e5",
        "libreswan": "4.12-r2",
        "luci-app-attendedsysupgrade": "25.103.51521~2ac26e5",
        "e2fsprogs": "1.47.0-r2",
        "kmod-e1000": "6.6.73-r1",
        "kmod-ppp": "6.6.73-r1",
        "ca-bundle": "20241223-r1",
        "kmod-forcedeth": "6.6.73-r1",
        "luci-i18n-base-zh-cn": "25.103.51521~2ac26e5",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "kmod-amd-xgbe": "6.6.73-r1",
        "luci-i18n-attendedsysupgrade-zh-cn": "25.103.51521~2ac26e5",
        "luci": "25.103.51521~2ac26e5",
        "luci-app-lucky": "2.1.1-1",
        "fdisk": "2.40.2-r1",
        "luci-light": "25.103.51521~2ac26e5",
        "kmod-lib-lzo": "6.6.73-r1",
        "luci-lib-uqr": "25.103.51521~2ac26e5",
        "kmod-amazon-ena": "6.6.73-r1",
        "kernel": "6.6.73~a21259e4f338051d27a6443a3a7f7f1f-r1",
        "kmod-e1000e": "6.6.73-r1",
        "luci-mod-network": "25.103.51521~2ac26e5",
        "kmod-button-hotplug": "6.6.73-r3",
        "liblucihttp-ucode": "2023.03.15~9b5b683f-r1",
        "iperf3": "3.17.1-r4",
        "kmod-igb": "6.6.73-r1",
        "kmod-igc": "6.6.73-r1",
        "kmod-bnx2": "6.6.73-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "luci-i18n-firewall-zh-cn": "25.103.51521~2ac26e5",
        "luci-lib-chartjs": "25.103.51521~2ac26e5",
        "ppp-mod-pppoe": "2.5.1-r1",
        "luci-i18n-nlbwmon-zh-cn": "25.103.51521~2ac26e5"
    },
    "profile": "qemu-standard-pc-q35-ich9-2009",
    "target": "x86/64",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "ext4"
}

proninyaroslav, 2025-04-15:
Same here:
```
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.035.62793~4f7a183",
    "packages": {
        "iwinfo": "2024.10.20~b94f066e-r1",
        "liblucihttp": "2023.03.15~9b5b683f-r1",
        "cgi-io": "2022.08.10~901b0f04-r21",
        "libc": "1.2.5-r4",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "iw": "6.9-r1",
        "kmod-crypto-aead": "6.6.73-r1",
        "luci-app-https-dns-proxy": "2023.12.26-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "ucode-mod-uloop": "2024.07.22~b610860d-r3",
        "getrandom": "2024.04.26~85f10530-r1",
        "kmod-phy-aquantia": "6.6.73-r1",
        "youtubeUnblock": "1.0.0-10-f37c3dd~f37c3dd-r10",
        "kmod-pppoe": "6.6.73-r1",
        "luci-lua-runtime": "25.035.62793~4f7a183",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "kmod-mt7986-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "uboot-envtools": "2024.07-r1",
        "dnsmasq": "2.90-r4",
        "kmod-crypto-acompress": "6.6.73-r1",
        "kmod-mt76-connac": "6.6.73.2025.01.14~8e4f72b6-r1",
        "libmbedtls": "3.6.2-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "luci-ssl": "25.035.62793~4f7a183",
        "dropbear": "2024.86-r1",
        "kmod-hwmon-core": "6.6.73-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "kmod-crypto-hw-safexcel": "6.6.73-r1",
        "kmod-crypto-authenc": "6.6.73-r1",
        "kmod-mt7915e": "6.6.73.2025.01.14~8e4f72b6-r1",
        "kmod-crypto-ccm": "6.6.73-r1",
        "urandom-seed": "3",
        "ppp": "2.5.1-r1",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-mac80211": "6.6.73.6.12.6-r1",
        "mt7986-wo-firmware": "20241110-r1",
        "eip197-mini-firmware": "20241110-r1",
        "luci-app-attendedsysupgrade": "25.035.62793~4f7a183",
        "kmod-libphy": "6.6.73-r1",
        "kmod-crypto-des": "6.6.73-r1",
        "luci-app-youtubeUnblock": "0.250220.77171",
        "kmod-ppp": "6.6.73-r1",
        "kmod-crypto-gcm": "6.6.73-r1",
        "ca-bundle": "20240203-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "kmod-crypto-cmac": "6.6.73-r1",
        "kmod-leds-ws2812b": "6.6.73-r1",
        "luci": "25.035.62793~4f7a183",
        "kmod-lib-lzo": "6.6.73-r1",
        "kmod-nft-queue": "6.6.73-r1",
        "kernel": "6.6.73~d649d775435da5a8637f7a955a80d331-r1",
        "kmod-nfnetlink-queue": "6.6.73-r1",
        "liblucihttp-ucode": "2023.03.15~9b5b683f-r1",
        "hostapd-common": "2024.09.15~5ace39b0-r2",
        "urngd": "2023.11.01~44365eb1-r1",
        "kmod-cfg80211": "6.6.73.6.12.6-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "xiaomi,redmi-router-ax6000-stock",
    "target": "mediatek/filogic",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "squashfs"
}
```

L01P, 2025-04-15:
Same here:

{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.103.51521~2ac26e5",
    "packages": {
        "libopenssl": "3.0.16-r1",
        "liblucihttp": "2023.03.15~9b5b683f-r1",
        "cgi-io": "2022.08.10~901b0f04-r21",
        "libc": "1.2.5-r4",
        "mkf2fs": "1.16.0-r3",
        "opkg": "2024.10.16~38eccbb1-r1",
        "kmod-ixgbe": "6.6.73-r1",
        "libiwinfo": "2024.10.20~b94f066e-r1",
        "bash": "5.2.32-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "resize2fs": "1.47.0-r2",
        "getrandom": "2024.04.26~85f10530-r1",
        "prometheus-node-exporter-lua-netstat": "2024.06.16-r2",
        "kmod-pppoe": "6.6.73-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "partx-utils": "2.40.2-r1",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "libf2fs": "1.16.0-r3",
        "libsmartcols": "2.40.2-r1",
        "kmod-r8169": "6.6.73-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "dnsmasq": "2.90-r4",
        "kmod-crypto-acompress": "6.6.73-r1",
        "prometheus-node-exporter-lua": "2024.06.16-r2",
        "tar": "1.35-r1",
        "kmod-fs-vfat": "6.6.73-r1",
        "grub2-efi": "2.12-r1",
        "luci-app-unbound": "25.103.51521~2ac26e5",
        "libmbedtls": "3.6.3-r1",
        "unbound-control": "1.21.0-r1",
        "grub2-bios-setup": "2.12-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "grub2": "2.12-r1",
        "losetup": "2.40.2-r1",
        "luci-ssl": "25.103.51521~2ac26e5",
        "dropbear": "2024.86-r1",
        "prometheus-node-exporter-lua-openwrt": "2024.06.16-r2",
        "adguardhome": "0.107.57-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "luci-app-sqm": "25.103.51521~2ac26e5",
        "parted": "3.6-r1",
        "urandom-seed": "3",
        "ppp": "2.5.1-r1",
        "socat": "1.8.0.0-r2",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-tg3": "6.6.73-r1",
        "qemu-ga": "9.1.0-r4",
        "luci-app-nft-qos": "25.103.51521~2ac26e5",
        "luci-app-attendedsysupgrade": "25.103.51521~2ac26e5",
        "prometheus-node-exporter-lua-nat_traffic": "2024.06.16-r2",
        "e2fsprogs": "1.47.0-r2",
        "kmod-e1000": "6.6.73-r1",
        "kmod-ppp": "6.6.73-r1",
        "netcat": "0.7.1-r2",
        "openssh-sftp-client": "9.9_p2-r1",
        "sshpass": "1.09-r1",
        "kmod-forcedeth": "6.6.73-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "kmod-amd-xgbe": "6.6.73-r1",
        "luci": "25.103.51521~2ac26e5",
        "kmod-lib-lzo": "6.6.73-r1",
        "kmod-amazon-ena": "6.6.73-r1",
        "kernel": "6.6.73~a21259e4f338051d27a6443a3a7f7f1f-r1",
        "kmod-e1000e": "6.6.73-r1",
        "kmod-button-hotplug": "6.6.73-r3",
        "liblucihttp-ucode": "2023.03.15~9b5b683f-r1",
        "kmod-igb": "6.6.73-r1",
        "kmod-igc": "6.6.73-r1",
        "kmod-bnx2": "6.6.73-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "luci-lib-chartjs": "25.103.51521~2ac26e5",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "qemu-standard-pc-i440fx-piix-1996",
    "target": "x86/64",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "ext4"
}