openwrt/asu #1325: Unsupported profile: qihoo,360t7-ubootmod.

g1fts, 2025-04-15:
Unsupported profile: qihoo,360t7-ubootmod. The requested profile was either dropped or never existed. Please check the forums for more information.

```bash
{
    "url": "https://sysupgrade.kyarucloud.moe",
    "branch": "23.05",
    "revision": "r28061-399f9a1db3",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/git-25.093.36456-7eaec9d",
    "packages": {
        "iwinfo": "2023-07-01-ca79f641-1",
        "libopenssl": "3.0.15-1",
        "liblucihttp": "2023-03-15-9b5b683f-1",
        "kmod-lib-textsearch": "5.15.167-1",
        "cgi-io": "2022-08-10-901b0f04-21",
        "luci-lib-base": "git-24.272.29284-d386ad6",
        "kmod-macvlan": "5.15.167-1",
        "kmod-crypto-user": "5.15.167-1",
        "kmod-crypto-gf128": "5.15.167-1",
        "luci-i18n-argon-config-zh-cn": "git-24.272.29284-d386ad6",
        "opkg": "2022-02-24-d038e5b6-2",
        "luci-app-opkg": "git-24.302.38724-3ded930",
        "ubus": "2023-06-05-f787c97b-1",
        "iw": "5.19-1",
        "kmod-crypto-manager": "5.15.167-1",
        "kmod-sched-act-vlan": "5.15.167-1",
        "libuci": "2023-08-10-5781664d-1",
        "kmod-nft-fullcone": "5.15.167+2023-01-10-95ad79bc-1",
        "rpcd": "2023-07-01-c07ab2f9-1",
        "busybox": "1.36.1-1",
        "kmod-crypto-ctr": "5.15.167-1",
        "libiwinfo": "2023-07-01-ca79f641-1",
        "luci-lib-ip": "git-24.272.29284-d386ad6",
        "kmod-nft-fib": "5.15.167-1",
        "kmod-nfnetlink": "5.15.167-1",
        "libubus-lua": "2023-06-05-f787c97b-1",
        "kmod-crypto-hash": "5.15.167-1",
        "kmod-nf-reject6": "5.15.167-1",
        "kmod-crypto-aead": "5.15.167-1",
        "kmod-crypto-md5": "5.15.167-1",
        "luci-mod-system": "git-24.365.26965-7378b30",
        "kmod-nf-flow": "5.15.167-1",
        "libnl-tiny": "2023-07-27-bc92a280-1",
        "kmod-lib-crc-ccitt": "5.15.167-1",
        "ucode-mod-uloop": "2024-07-11-1a8a0bcf-3",
        "getrandom": "2022-08-13-4c7b720b-2",
        "ucode-mod-ubus": "2024-07-11-1a8a0bcf-3",
        "kmod-phy-aquantia": "5.15.167-1",
        "kmod-asn1-decoder": "5.15.167-1",
        "libustream-openssl": "2023-02-25-498f6e26-1",
        "luci-theme-bootstrap": "git-24.272.29284-d386ad6",
        "kmod-nf-nathelper": "5.15.167-1",
        "kmod-pppoe": "5.15.167-1",
        "kmod-sched-bpf": "5.15.167-1",
        "luci-lua-runtime": "git-24.272.29284-d386ad6",
        "kmod-pppox": "5.15.167-1",
        "shellsync": "0.2-2",
        "kmod-nf-reject": "5.15.167-1",
        "procd-ujail": "2023-06-25-2db83655-2",
        "base-files": "1721-r28061-399f9a1db3",
        "kmod-nf-nat": "5.15.167-1",
        "kmod-crypto-crc32c": "5.15.167-1",
        "ucode-mod-uci": "2024-07-11-1a8a0bcf-3",
        "ucode-mod-lua": "1",
        "netifd": "2024-01-04-c18cc79d-2",
        "6in4": "28",
        "firewall4": "2023-09-01-598d9fbb-1",
[... 128 lines cut ...]
        "kmod-sched-flower": "5.15.167-1",
        "jsonfilter": "2024-01-23-594cfa86-1",
        "liblucihttp-ucode": "2023-03-15-9b5b683f-1",
        "libubox": "2023-05-23-75a3b870-1",
        "hostapd-common": "2023-09-08-e5ccbfc6-8",
        "kmod-crypto-arc4": "5.15.167-1",
        "kmod-crypto-sha1": "5.15.167-1",
        "wireless-regdb": "2024.10.07-1",
        "urngd": "2023-11-01-44365eb1-1",
        "kmod-slhc": "5.15.167-1",
        "kmod-cfg80211": "5.15.167+6.1.110-1-1",
        "luci-i18n-firewall-zh-cn": "git-25.093.36456-7eaec9d",
        "rpcd-mod-rrdns": "20170710",
        "ppp-mod-pppoe": "2.4.9.git-2021-01-04-7",
        "luci-app-passwall": "25.4.1-1",
        "kmod-sched-core": "5.15.167-1",
        "dnsmasq-full": "2.90-2"
    },
    "profile": "qihoo,360t7-ubootmod",
    "target": "mediatek/filogic",
    "version": "23.05.4",
    "diff_packages": true,
    "filesystem": "squashfs"
}
```

dc366, 2025-04-18:
You are starting from a firmware which is not an official OpenWRT release, which causes `asu` to fail.

g1fts, 2025-04-24:
> You are starting from a firmware which is not an official OpenWRT release, which causes `asu` to fail.

ok, got it

closed 2025-04-24 by g1fts