openwrt/asu #1333: Unsupported version: 24.10.1 attempting "Attended Sysupgrade" from 24.10.0 (r28427-6df0e3d02a) on Linksys E8450 (UBI)

gene-hightower, 2025-04-15:
Error building the firmware image

Server response: Unsupported version: 24.10.1
Please report the error message and request

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.035.62793~4f7a183",
    "packages": {
        "unbound-host": "1.21.0-r1",
        "libc": "1.2.5-r4",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "libpthread": "1.2.5-r4",
        "odhcpd": "2024.05.08~a2988231-r1",
        "kmod-crypto-aead": "6.6.73-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "kmod-pppoe": "6.6.73-r1",
        "kmod-mt7622-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "unbound-daemon": "1.21.0-r1",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "fitblk": "2",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "uboot-envtools": "2024.07-r1",
        "dnsmasq": "2.90-r4",
        "kmod-mt7615-common": "6.6.73.2025.01.14~8e4f72b6-r1",
        "kmod-crypto-acompress": "6.6.73-r1",
        "kmod-mt76-connac": "6.6.73.2025.01.14~8e4f72b6-r1",
        "kmod-usb3": "6.6.73-r1",
        "luci-app-unbound": "25.044.01357~67d27ad",
        "libmbedtls": "3.6.2-r1",
        "unbound-control": "1.21.0-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "luci-ssl": "25.035.62793~4f7a183",
        "dropbear": "2024.86-r1",
        "kmod-hwmon-core": "6.6.73-r1",
        "mtd": "26",
        "kmod-crypto-ccm": "6.6.73-r1",
        "urandom-seed": "3",
        "ppp": "2.5.1-r1",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-mac80211": "6.6.73.6.12.6-r1",
        "luci-app-attendedsysupgrade": "25.035.62793~4f7a183",
        "kmod-ppp": "6.6.73-r1",
        "kmod-mt7915-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "kmod-crypto-gcm": "6.6.73-r1",
        "ca-bundle": "20240203-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "kmod-crypto-cmac": "6.6.73-r1",
        "bind-dig": "9.20.4-r1",
        "luci": "25.035.62793~4f7a183",
        "kmod-lib-lzo": "6.6.73-r1",
        "kernel": "6.6.73~49f3b78722ed2df9df713e744e978eb6-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "kmod-cfg80211": "6.6.73.6.12.6-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "linksys,e8450-ubi",
    "target": "mediatek/mt7622",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "squashfs"
}

c3v1t, 2025-04-15:
Hello, I have the same problem
Request data:
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.103.51521~2ac26e5",
    "packages": {
        "luci-theme-material": "25.103.51521~2ac26e5",
        "luci-app-statistics": "25.103.51521~2ac26e5",
        "banip": "1.5.6-r1",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "libpthread": "1.2.5-r4",
        "rpcd": "2024.09.17~9f4b86e7-r1",
        "kmod-nft-fib": "6.6.73-r1",
        "kmod-nfnetlink": "6.6.73-r1",
        "kmod-nf-reject6": "6.6.73-r1",
        "nano": "8.4-r1",
        "kmod-crypto-aead": "6.6.73-r1",
        "luci-mod-system": "25.103.51521~2ac26e5",
        "kmod-nf-flow": "6.6.73-r1",
        "libnl-tiny": "2025.03.19~c0df580a-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "ucode-mod-uloop": "2025.02.10~a8a11aea-r1",
        "px5g-mbedtls": "11",
        "getrandom": "2024.04.26~85f10530-r1",
        "ucode-mod-ubus": "2025.02.10~a8a11aea-r1",
        "luci-theme-bootstrap": "25.103.51521~2ac26e5",
        "kmod-pppoe": "6.6.73-r1",
        "kmod-pppox": "6.6.73-r1",
        "kmod-mt7622-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "kmod-nf-reject": "6.6.73-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "kmod-nf-nat": "6.6.73-r1",
        "luci-nginx": "25.103.51521~2ac26e5",
        "kmod-crypto-crc32c": "6.6.73-r1",
        "ucode-mod-uci": "2025.02.10~a8a11aea-r1",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "fitblk": "2",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "uboot-envtools": "2024.07-r1",
        "dnsmasq": "2.90-r4",
        "kmod-mt7615-common": "6.6.73.2025.01.14~8e4f72b6-r1",
        "luci-app-banip": "25.103.51521~2ac26e5",
        "rpcd-mod-ucode": "2024.09.17~9f4b86e7-r1",
        "kmod-crypto-acompress": "6.6.73-r1",
        "ucode-mod-math": "2025.02.10~a8a11aea-r1",
        "kmod-lib-crc32c": "6.6.73-r1",
        "luci-mod-status": "25.103.51521~2ac26e5",
        "collectd-mod-cpufreq": "5.12.0-r53",
        "kmod-mt76-connac": "6.6.73.2025.01.14~8e4f72b6-r1",
        "kmod-nft-nat": "6.6.73-r1",
        "luci-app-ddns": "25.103.51521~2ac26e5",
        "kmod-usb3": "6.6.73-r1",
        "luci-app-firewall": "25.103.51521~2ac26e5",
        "libmbedtls": "3.6.3-r1",
        "libmnl": "1.0.5-r1",
        "jansson": "2.14-r3",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uclient-fetch": "2024.10.22~88ae8f20-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "ucode-mod-bpf": "1",
        "ucode-mod-fs": "2025.02.10~a8a11aea-r1",
        "collectd-mod-nginx": "5.12.0-r53",
        "luci-ssl": "25.103.51521~2ac26e5",
        "dropbear": "2024.86-r1",
        "kmod-hwmon-core": "6.6.73-r1",
        "libnftnl": "1.2.8-r1",
        "rpcd-mod-file": "2024.09.17~9f4b86e7-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "collectd-mod-ipstatistics": "5.12.0-r53",
        "ucode-mod-nl80211": "2025.02.10~a8a11aea-r1",
        "ucode": "2025.02.10~a8a11aea-r1",
        "kmod-crypto-ccm": "6.6.73-r1",
        "luci-app-sqm": "25.103.51521~2ac26e5",
        "kmod-nf-log": "6.6.73-r1",
        "luci-app-package-manager": "25.103.51521~2ac26e5",
        "urandom-seed": "3",
        "luci-proto-ppp": "25.103.51521~2ac26e5",
        "ppp": "2.5.1-r1",
        "luci-mod-admin-full": "25.103.51521~2ac26e5",
        "luci-base": "25.103.51521~2ac26e5",
        "bpftool-full": "7.5.0-r1",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-nf-log6": "6.6.73-r1",
        "kmod-mac80211": "6.6.73.6.12.6-r1",
        "luci-proto-ipv6": "25.103.51521~2ac26e5",
        "464xlat": "13",
        "luci-app-attendedsysupgrade": "25.103.51521~2ac26e5",
        "kmod-nft-offload": "6.6.73-r1",
        "kmod-ppp": "6.6.73-r1",
        "rrdcgi1": "1.0.50-r6",
        "libpcap": "1.10.5-r2",
        "uhttpd": "2023.06.25~34a8a74d-r4",
        "kmod-nf-conntrack": "6.6.73-r1",
        "device-observatory": "1.2.0-r7",
        "collectd-mod-exec": "5.12.0-r53",
        "kmod-mt7915-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "kmod-crypto-gcm": "6.6.73-r1",
        "ucode-mod-rtnl": "2025.02.10~a8a11aea-r1",
        "ca-bundle": "20241223-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "kmod-crypto-cmac": "6.6.73-r1",
        "luci-app-bcp38": "25.103.51521~2ac26e5",
        "luci": "25.103.51521~2ac26e5",
        "kmod-nf-conntrack6": "6.6.73-r1",
        "luci-light": "25.103.51521~2ac26e5",
        "kmod-lib-lzo": "6.6.73-r1",
        "kmod-mt76-core": "6.6.73.2025.01.14~8e4f72b6-r1",
        "libpfring": "8.6.1-r1",
        "rpcd-mod-iwinfo": "2024.09.17~9f4b86e7-r1",
        "luci-mod-network": "25.103.51521~2ac26e5",
        "kmod-nft-core": "6.6.73-r1",
        "libucode": "2025.02.10~a8a11aea-r1",
        "smartdns": "1.2023.43-r1",
        "uhttpd-mod-ubus": "2023.06.25~34a8a74d-r4",
        "luci-app-smartdns": "1.2023.42-r1",
        "collectd-mod-ethstat": "5.12.0-r53",
        "collectd-mod-dns": "5.12.0-r53",
        "collectd-mod-protocols": "5.12.0-r53",
        "kmod-bpf-test": "6.6.73-r1",
        "wireless-regdb": "2025.02.20-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "kmod-slhc": "6.6.73-r1",
        "kmod-cfg80211": "6.6.73.6.12.6-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "linksys,e8450-ubi",
    "target": "mediatek/mt7622",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "squashfs"
}

EygenCat, 2025-04-15:
```
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.103.51521~2ac26e5",
    "packages": {
        "libc": "1.2.5-r4",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "libpthread": "1.2.5-r4",
        "kmod-crypto-aead": "6.6.73-r1",
        "luci-mod-system": "25.103.51521~2ac26e5",
        "libnl-tiny": "2025.03.19~c0df580a-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "ucode-mod-uloop": "2025.02.10~a8a11aea-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "luci-theme-bootstrap": "25.103.51521~2ac26e5",
        "kmod-pppoe": "6.6.73-r1",
        "kmod-mt7622-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "fitblk": "2",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "uboot-envtools": "2024.07-r1",
        "dnsmasq": "2.90-r4",
        "kmod-mt7615-common": "6.6.73.2025.01.14~8e4f72b6-r1",
        "kmod-crypto-acompress": "6.6.73-r1",
        "luci-mod-status": "25.103.51521~2ac26e5",
        "luci-i18n-base-ru": "25.103.51521~2ac26e5",
        "kmod-mt76-connac": "6.6.73.2025.01.14~8e4f72b6-r1",
        "luci-app-firewall": "25.103.51521~2ac26e5",
        "libmbedtls": "3.6.3-r1",
        "luci-i18n-attendedsysupgrade-ru": "25.103.51521~2ac26e5",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "luci-ssl": "25.103.51521~2ac26e5",
        "dropbear": "2024.86-r1",
        "kmod-hwmon-core": "6.6.73-r1",
        "adguardhome": "0.107.57-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "kmod-crypto-ccm": "6.6.73-r1",
        "luci-app-package-manager": "25.103.51521~2ac26e5",
        "urandom-seed": "3",
        "luci-proto-ppp": "25.103.51521~2ac26e5",
        "ppp": "2.5.1-r1",
        "luci-mod-admin-full": "25.103.51521~2ac26e5",
        "luci-base": "25.103.51521~2ac26e5",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-mac80211": "6.6.73.6.12.6-r1",
        "luci-proto-ipv6": "25.103.51521~2ac26e5",
        "luci-app-attendedsysupgrade": "25.103.51521~2ac26e5",
        "attendedsysupgrade-common": "9",
        "kmod-ppp": "6.6.73-r1",
        "luci-i18n-package-manager-ru": "25.103.51521~2ac26e5",
        "luci-proto-relay": "25.103.51521~2ac26e5",
        "kmod-mt7915-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "kmod-crypto-gcm": "6.6.73-r1",
        "ca-bundle": "20241223-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "kmod-crypto-cmac": "6.6.73-r1",
        "luci": "25.103.51521~2ac26e5",
        "luci-light": "25.103.51521~2ac26e5",
        "kmod-lib-lzo": "6.6.73-r1",
        "kernel": "6.6.73~49f3b78722ed2df9df713e744e978eb6-r1",
        "luci-mod-network": "25.103.51521~2ac26e5",
        "urngd": "2023.11.01~44365eb1-r1",
        "kmod-cfg80211": "6.6.73.6.12.6-r1",
        "ppp-mod-pppoe": "2.5.1-r1",
        "luci-i18n-firewall-ru": "25.103.51521~2ac26e5"
    },
    "profile": "xiaomi,redmi-router-ax6s",
    "target": "mediatek/mt7622",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "squashfs"
}
```

gene-hightower, 2025-04-18:
After removing odhcpd-ipv6only (as I also had odhcpd installed) and retrying the update, it worked.

closed 2025-04-18 by gene-hightower