openwrt/asu #1335: Server response: Unsupported version: 24.10.1 - ASUS RT-AX54

noobTrying2learn, 2025-04-16:
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.104.48032~8e5fed1",
    "packages": {
        "unbound-host": "1.21.0-r1",
        "iwinfo": "2024.10.20~b94f066e-r1",
        "liblucihttp": "2023.03.15~9b5b683f-r1",
        "luci-app-adblock": "25.104.48032~8e5fed1",
        "cgi-io": "2022.08.10~901b0f04-r21",
        "luci-lib-base": "25.104.48032~8e5fed1",
        "libc": "1.2.5-r4",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "luci-lib-ip": "25.104.48032~8e5fed1",
        "luci-mod-system": "25.104.48032~8e5fed1",
        "libnl-tiny": "2025.03.19~c0df580a-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "ucode-mod-uloop": "2025.02.10~a8a11aea-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "unbound-checkconf": "1.21.0-r1",
        "luci-theme-bootstrap": "25.104.48032~8e5fed1",
        "kmod-pppoe": "6.6.73-r1",
        "luci-lua-runtime": "25.104.48032~8e5fed1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "uboot-envtools": "2024.07-r1",
        "adblock": "4.2.7-r3",
        "dnsmasq": "2.90-r4",
        "kmod-crypto-acompress": "6.6.73-r1",
        "luci-mod-status": "25.104.48032~8e5fed1",
        "https-dns-proxy": "2025.03.13-r1",
        "luci-app-firewall": "25.104.48032~8e5fed1",
        "luci-app-unbound": "25.104.48032~8e5fed1",
        "libmbedtls": "3.6.3-r1",
        "unbound-control-setup": "1.21.0-r1",
        "luci-lib-ipkg": "25.104.48032~8e5fed1",
        "unbound-control": "1.21.0-r1",
        "tcpdump": "4.99.5-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "luci-ssl": "25.104.48032~8e5fed1",
        "dropbear": "2024.86-r1",
        "luci-theme-argon": "2.3.2-r20250207",
        "luci-compat": "25.104.48032~8e5fed1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "luci-app-package-manager": "25.104.48032~8e5fed1",
        "urandom-seed": "3",
        "luci-proto-ppp": "25.104.48032~8e5fed1",
        "ppp": "2.5.1-r1",
        "luci-mod-admin-full": "25.104.48032~8e5fed1",
        "luci-base": "25.104.48032~8e5fed1",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "luci-proto-ipv6": "25.104.48032~8e5fed1",
        "luci-app-attendedsysupgrade": "25.104.48032~8e5fed1",
        "kmod-ppp": "6.6.73-r1",
        "git": "2.46.2-r1",
        "kmod-mt7915-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "htop": "3.3.0-r2",
        "luci-lib-nixio": "25.104.48032~8e5fed1",
        "ca-bundle": "20241223-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "luci-lib-jsonc": "25.104.48032~8e5fed1",
        "luci-app-argon-config": "0.9",
        "luci": "25.104.48032~8e5fed1",
        "owut": "2025.03.14~52e7d44c-r1",
        "luci-light": "25.104.48032~8e5fed1",
        "kmod-lib-lzo": "6.6.73-r1",
        "kernel": "6.6.73~3abe85def815b59c6c75ac1f92135cb6-r1",
        "unbound-anchor": "1.21.0-r1",
        "luci-mod-network": "25.104.48032~8e5fed1",
        "liblucihttp-ucode": "2023.03.15~9b5b683f-r1",
        "hostapd-common": "2024.09.15~5ace39b0-r2",
        "iperf3": "3.17.1-r4",
        "urngd": "2023.11.01~44365eb1-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "asus,rt-ax54",
    "target": "ramips/mt7621",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "squashfs"
}

noobTrying2learn, 2025-04-17:
My apologies as I'm sure you're quite busy. As my name implies, I'm new to this and am trying to learn. If you would be so kind to why the thumbs down and the multitude of stuck-through lines was your response, perhaps I can learn something. 

As of now I'm baffled. My router is functioning, all packages have been downloaded and installed via opkg. I submitted the initial information as instructed following an error when attempting to upgrade via the attended sysupgrade function.

Sorry for what is likely a stupid question. 

Regards,
B

robimarko, 2025-04-17:
Just ignore the thumbs down, anybody can do the thumbs up/down reaction.

As for the strikethrough lines, nobody replied to you, that was your initial post which you needed to enclose with a code block so that GH does not interpret it as markdown

noobTrying2learn, 2025-04-21:
Ok, thank you for the explanation. I appreciate it.