openwrt/asu #1341: Attended Sysupgrade for openwrt one not working in release 24.10.1

peterb99, 2025-04-16:
Upgrade from 14.10 to 24.10.1
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.104.48032~8e5fed1",
    "packages": {
        "kmod-usb-storage": "6.6.73-r1",
        "iwinfo": "2024.10.20~b94f066e-r1",
        "kmod-ip6tables-extra": "6.6.73-r1",
        "luci-app-statistics": "25.104.48032~8e5fed1",
        "luci-app-adblock": "25.104.48032~8e5fed1",
        "libc": "1.2.5-r4",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "kmod-usb-storage-uas": "6.6.73-r1",
        "diffutils": "3.10-r1",
        "kmod-crypto-aead": "6.6.73-r1",
        "ppp-mod-radius": "2.5.1-r1",
        "luci-app-aria2": "25.104.48032~8e5fed1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "kmod-phy-aquantia": "6.6.73-r1",
        "kmod-pppoe": "6.6.73-r1",
        "speedtest-go": "1.7.7-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "collectd-mod-match-timediff": "5.12.0-r53",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "fitblk": "2",
        "ddns-scripts": "2.8.2-r64",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "adblock": "4.2.7-r3",
        "uboot-envtools": "2024.07-r1",
        "dnsmasq": "2.90-r4",
        "kmod-crypto-hmac": "6.6.73-r1",
        "kmod-nvme": "6.6.73-r1",
        "kmod-crypto-acompress": "6.6.73-r1",
        "kmod-rtc-pcf8563": "6.6.73-r1",
        "collectd-mod-cpufreq": "5.12.0-r53",
        "block-mount": "2024.07.14~408c2cc4-r1",
        "luci-app-ddns": "25.104.48032~8e5fed1",
        "kmod-usb2": "6.6.73-r1",
        "kmod-usb3": "6.6.73-r1",
        "libmbedtls": "3.6.3-r1",
        "tcpdump": "4.99.5-r1",
        "ethtool-full": "6.11-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "fstrim": "2.40.2-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "kmod-fs-ext4": "6.6.73-r1",
        "collectd": "5.12.0-r53",
        "luci-ssl": "25.104.48032~8e5fed1",
        "dropbear": "2024.86-r1",
        "logrotate": "3.22.0-r1",
        "kmod-hwmon-core": "6.6.73-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "collectd-mod-ipstatistics": "5.12.0-r53",
        "kmod-crypto-hw-safexcel": "6.6.73-r1",
        "kmod-crypto-authenc": "6.6.73-r1",
        "kmod-mt7915e": "6.6.73.2025.01.14~8e4f72b6-r1",
        "urandom-seed": "3",
        "ppp": "2.5.1-r1",
        "mbedtls-util": "3.6.3-r1",
        "luci-app-commands": "25.104.48032~8e5fed1",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "luci-app-attendedsysupgrade": "25.104.48032~8e5fed1",
        "collectd-mod-dhcpleases": "5.12.0-r53",
        "kmod-libphy": "6.6.73-r1",
        "kmod-crypto-des": "6.6.73-r1",
        "e2fsprogs": "1.47.0-r2",
        "attendedsysupgrade-common": "9",
        "pppdump": "2.5.1-r1",
        "ca-certificates": "20241223-r1",
        "kmod-ppp": "6.6.73-r1",
        "samba4-utils": "4.18.8-r1",
        "collectd-mod-ntpd": "5.12.0-r53",
        "luci-app-samba4": "25.104.48032~8e5fed1",
        "avahi-utils": "0.8-r9",
        "mt7981-wo-firmware": "20241110-r1",
        "dhcpcd": "10.0.10-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "luci": "25.104.48032~8e5fed1",
        "samba4-admin": "4.18.8-r1",
        "owut": "2025.03.14~52e7d44c-r1",
        "fdisk": "2.40.2-r1",
        "kmod-mt7981-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "kmod-lib-lzo": "6.6.73-r1",
        "kernel": "6.6.73~d649d775435da5a8637f7a955a80d331-r1",
        "kmod-phy-airoha-en8811h": "6.6.73-r1",
        "aria2": "1.37.0-r1",
        "luci-app-acl": "25.104.48032~8e5fed1",
        "collectd-mod-dns": "5.12.0-r53",
        "hostapd-common": "2024.09.15~5ace39b0-r2",
        "collectd-mod-thermal": "5.12.0-r53",
        "luci-app-hd-idle": "25.104.48032~8e5fed1",
        "urngd": "2023.11.01~44365eb1-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "openwrt,one",
    "target": "mediatek/filogic",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "squashfs"

dc366, 2025-04-18:
Are you really trying to upgrade from 14.10 to 24.10? That's not going to work.

peterb99, 2025-04-18:
sorry, typo. Problem seems to have been solved already. Great work. Much appreciated