openwrt/asu #1344: Attended Sysupgrade : Unsupported version: 24.10.1 for GLinet MT-6000

Uncensored-Hosting, 2025-04-16:
Error building the firmware image  Server response: Unsupported version: 24.10.1

goodness-from-me, 2025-04-16:
Adding more info to the issue:

```
Error building the firmware image

Server response: Unsupported version: 24.10.1
```

Request Data:
```json
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.037.68331~42f464c",
    "packages": {
        "iwinfo": "2024.10.20~b94f066e-r1",
        "liblucihttp": "2023.03.15~9b5b683f-r1",
        "cgi-io": "2022.08.10~901b0f04-r21",
        "libc": "1.2.5-r4",
        "mkf2fs": "1.16.0-r3",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "iw": "6.9-r1",
        "kmod-crypto-aead": "6.6.73-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "ucode-mod-uloop": "2024.07.22~b610860d-r3",
        "getrandom": "2024.04.26~85f10530-r1",
        "kmod-phy-aquantia": "6.6.73-r1",
        "kmod-pppoe": "6.6.73-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "kmod-mt7986-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "uboot-envtools": "2024.07-r1",
        "dnsmasq": "2.90-r4",
        "kmod-crypto-acompress": "6.6.73-r1",
        "kmod-mt76-connac": "6.6.73.2025.01.14~8e4f72b6-r1",
        "kmod-usb3": "6.6.73-r1",
        "libmbedtls": "3.6.2-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "luci-ssl": "25.037.68331~42f464c",
        "dropbear": "2024.86-r1",
        "kmod-hwmon-core": "6.6.73-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "kmod-crypto-hw-safexcel": "6.6.73-r1",
        "kmod-crypto-authenc": "6.6.73-r1",
        "kmod-mt7915e": "6.6.73.2025.01.14~8e4f72b6-r1",
        "kmod-crypto-ccm": "6.6.73-r1",
        "urandom-seed": "3",
        "ppp": "2.5.1-r1",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-mac80211": "6.6.73.6.12.6-r1",
        "mt7986-wo-firmware": "20241110-r1",
        "eip197-mini-firmware": "20241110-r1",
        "luci-app-attendedsysupgrade": "25.037.68331~42f464c",
        "kmod-libphy": "6.6.73-r1",
        "kmod-crypto-des": "6.6.73-r1",
        "e2fsprogs": "1.47.0-r2",
        "kmod-ppp": "6.6.73-r1",
        "kmod-crypto-gcm": "6.6.73-r1",
        "f2fsck": "1.16.0-r3",
        "ca-bundle": "20240203-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "kmod-crypto-cmac": "6.6.73-r1",
        "luci": "25.037.68331~42f464c",
        "kmod-lib-lzo": "6.6.73-r1",
        "kernel": "6.6.73~d649d775435da5a8637f7a955a80d331-r1",
        "liblucihttp-ucode": "2023.03.15~9b5b683f-r1",
        "hostapd-common": "2024.09.15~5ace39b0-r2",
        "urngd": "2023.11.01~44365eb1-r1",
        "kmod-cfg80211": "6.6.73.6.12.6-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "glinet,gl-mt6000",
    "target": "mediatek/filogic",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "squashfs"
}
```

goodness-from-me, 2025-04-16:
See #1327

closed 2025-04-17 by robimarko