openwrt/asu #1351: Error building the firmware image - Server Response: init -Linksys EA7500 V2

Jojje1999, 2025-04-17:
Issue is also present when using firmware selector if you add "luci-app-attendedsysupgrade" to installed packages when you request the build.

{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.035.62793~4f7a183",
    "packages": {
        "iwinfo": "2024.10.20~b94f066e-r1",
        "liblucihttp": "2023.03.15~9b5b683f-r1",
        "cgi-io": "2022.08.10~901b0f04-r21",
        "libc": "1.2.5-r4",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "ucode-mod-uloop": "2024.07.22~b610860d-r3",
        "getrandom": "2024.04.26~85f10530-r1",
        "kmod-pppoe": "6.6.73-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "uboot-envtools": "2024.07-r1",
        "dnsmasq": "2.90-r4",
        "kmod-crypto-acompress": "6.6.73-r1",
        "kmod-usb3": "6.6.73-r1",
        "libmbedtls": "3.6.2-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "luci-ssl": "25.035.62793~4f7a183",
        "dropbear": "2024.86-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "urandom-seed": "3",
        "ppp": "2.5.1-r1",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "luci-app-attendedsysupgrade": "25.035.62793~4f7a183",
        "kmod-ppp": "6.6.73-r1",
        "kmod-mt7615-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "ca-bundle": "20240203-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "luci": "25.035.62793~4f7a183",
        "kmod-lib-lzo": "6.6.73-r1",
        "kernel": "6.6.73~3abe85def815b59c6c75ac1f92135cb6-r1",
        "liblucihttp-ucode": "2023.03.15~9b5b683f-r1",
        "hostapd-common": "2024.09.15~5ace39b0-r2",
        "urngd": "2023.11.01~44365eb1-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "linksys,ea7500-v2",
    "target": "ramips/mt7621",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "squashfs"
}

dc366, 2025-04-17:
Try again, the imagebuilder container should be available now.

Jojje1999, 2025-04-17:
> Try again, the imagebuilder container should be available now.

Thank you for looking into this! I've now successfully updated my device.

closed 2025-04-17 by Jojje1999