openwrt/asu #1352: Error building the firmware image - Server Response: init - Asus RT-AX54

Imlbrown, 2025-04-17:
[
Currently running: 24.10.0 - r28427-6df0e3d02a

Search for firmware upgrade

Powered by [LuCI openwrt-24.10 branch (25.037.68331~42f464c)](https://github.com/openwrt/luci) / [OpenWrt 24.10.0 (r28427-6df0e3d02a)](https://openwrt.org/)
Error building the firmware image
Server response: init

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.053.28518~23ed1e5",
    "packages": {
        "iwinfo": "2024.10.20~b94f066e-r1",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "libpthread": "1.2.5-r4",
        "rpcd": "2024.09.17~9f4b86e7-r1",
        "libiwinfo": "2024.10.20~b94f066e-r1",
        "kmod-nft-fib": "6.6.73-r1",
        "kmod-nfnetlink": "6.6.73-r1",
        "kmod-nf-reject6": "6.6.73-r1",
        "kmod-crypto-aead": "6.6.73-r1",
        "luci-mod-system": "25.037.68331~42f464c",
        "kmod-nf-flow": "6.6.73-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "px5g-mbedtls": "11",
        "getrandom": "2024.04.26~85f10530-r1",
        "luci-theme-bootstrap": "25.037.68331~42f464c",
        "kmod-pppoe": "6.6.73-r1",
        "kmod-pppox": "6.6.73-r1",
        "kmod-nf-reject": "6.6.73-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "kmod-nf-nat": "6.6.73-r1",
        "kmod-crypto-crc32c": "6.6.73-r1",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "uboot-envtools": "2024.07-r1",
        "dnsmasq": "2.90-r4",
        "rpcd-mod-ucode": "2024.09.17~9f4b86e7-r1",
        "kmod-crypto-acompress": "6.6.73-r1",
        "ucode-mod-math": "2024.07.22~b610860d-r3",
        "kmod-lib-crc32c": "6.6.73-r1",
        "luci-mod-status": "25.037.68331~42f464c",
        "kmod-nft-nat": "6.6.73-r1",
        "luci-app-firewall": "25.037.68331~42f464c",
        "libmbedtls": "3.6.2-r1",
        "libmnl": "1.0.5-r1",
        "jansson": "2.14-r3",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uclient-fetch": "2024.10.22~88ae8f20-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "luci-ssl": "25.037.68331~42f464c",
        "dropbear": "2024.86-r1",
        "libnftnl": "1.2.8-r1",
        "rpcd-mod-file": "2024.09.17~9f4b86e7-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "kmod-crypto-authenc": "6.6.73-r1",
        "kmod-nf-log": "6.6.73-r1",
        "luci-app-package-manager": "25.037.68331~42f464c",
        "urandom-seed": "3",
        "luci-proto-ppp": "25.037.68331~42f464c",
        "ppp": "2.5.1-r1",
        "luci-mod-admin-full": "25.037.68331~42f464c",
        "luci-base": "25.037.68331~42f464c",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-nf-log6": "6.6.73-r1",
        "luci-proto-ipv6": "25.037.68331~42f464c",
        "luci-app-attendedsysupgrade": "25.053.28518~23ed1e5",
        "kmod-crypto-des": "6.6.73-r1",
        "kmod-crypto-hw-eip93": "6.6.73-r1",
        "kmod-nft-offload": "6.6.73-r1",
        "kmod-ppp": "6.6.73-r1",
        "uhttpd": "2023.06.25~34a8a74d-r4",
        "kmod-nf-conntrack": "6.6.73-r1",
        "kmod-mt7915-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "ca-bundle": "20240203-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "luci": "25.053.28518~23ed1e5",
        "kmod-nf-conntrack6": "6.6.73-r1",
        "luci-light": "25.037.68331~42f464c",
        "kmod-lib-lzo": "6.6.73-r1",
        "rpcd-mod-iwinfo": "2024.09.17~9f4b86e7-r1",
        "luci-mod-network": "25.037.68331~42f464c",
        "kmod-nft-core": "6.6.73-r1",
        "libucode": "2025.02.10~a8a11aea-r1",
        "uhttpd-mod-ubus": "2023.06.25~34a8a74d-r4",
        "hostapd-common": "2024.09.15~5ace39b0-r2",
        "urngd": "2023.11.01~44365eb1-r1",
        "kmod-slhc": "6.6.73-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "asus,rt-ax54",
    "target": "ramips/mt7621",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "squashfs"
}

dc366, 2025-04-17:
Try again, it looks like the imagebuilder container should be there now.

Imlbrown, 2025-04-17:
That fixed it, thanks!

closed 2025-04-17 by Imlbrown