openwrt/asu #1353: Error building the firmware image

hipyskipy, 2025-04-17:
Server response: init

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "ea1b7bcd05e0759215163bee20f9e8457897abe1c09e52660a1a44b7da2d4d74",
    "sha256_unsigned": "",
    "client": "luci/25.103.51521~2ac26e5",
    "packages": {
        "iwinfo": "2024.10.20~b94f066e-r1",
        "liblucihttp": "2023.03.15~9b5b683f-r1",
        "cgi-io": "2022.08.10~901b0f04-r21",
        "luci-app-adblock": "25.103.51521~2ac26e5",
        "libc": "1.2.5-r4",
        "luci-app-nlbwmon": "25.103.51521~2ac26e5",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "nano": "8.4-r1",
        "luci-app-https-dns-proxy": "2023.12.26-r4",
        "avahi-daemon-service-http": "0.8-r9",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "bind-host": "9.20.4-r1",
        "kmod-pppoe": "6.6.73-r1",
        "cifsmount": "6.11-r2",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "dnsmasq": "2.90-r4",
        "kmod-crypto-acompress": "6.6.73-r1",
        "block-mount": "2024.07.14~408c2cc4-r1",
        "luci-app-ddns": "25.103.51521~2ac26e5",
        "kmod-usb3": "6.6.73-r1",
        "kmod-fs-nfs-v4": "6.6.73-r1",
        "tcpdump": "4.99.5-r1",
        "nfs-utils": "2.6.2-r4",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "ethtool": "6.11-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "luci-ssl": "25.103.51521~2ac26e5",
        "dropbear": "2024.86-r1",
        "curl": "8.10.1-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "luci-app-sqm": "25.103.51521~2ac26e5",
        "urandom-seed": "3",
        "ppp": "2.5.1-r1",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "luci-app-attendedsysupgrade": "25.103.51521~2ac26e5",
        "irqbalance": "1.9.4-r1",
        "ca-certificates": "20241223-r1",
        "kmod-mt7615-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "htop": "3.3.0-r2",
        "luci": "25.103.51521~2ac26e5",
        "owut": "2025.03.14~52e7d44c-r1",
        "kmod-usb-ledtrig-usbport": "6.6.73-r1",
        "kmod-lib-lzo": "6.6.73-r1",
        "kernel": "6.6.73~3abe85def815b59c6c75ac1f92135cb6-r1",
        "iperf3": "3.17.1-r4",
        "urngd": "2023.11.01~44365eb1-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "dlink,dir-2640-a1",
    "target": "ramips/mt7621",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "squashfs"
}
<snip>

owut upgrade on same device:
Target         ramips/mt7621
Profile        dlink_dir-2640-a1
Package-arch   mipsel_24kc

yields:

Status:   init
Progress:   0s total =   0s in queue +   0s in build

Build failed in   0s total =   0s in queue +   0s to build:
ASU server error =
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/rq/worker.py", line 1633, in perform_job
    return_value = job.perform()
                   ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/rq/job.py", line 1331, in perform
    self._result = self._execute()
                   ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/rq/job.py", line 1365, in _execute
    result = self.func(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/asu/build.py", line 159, in build
    container = podman.containers.create(
                ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/podman/domain/containers_create.py", line 354, in create
    response.raise_for_status(not_found=ImageNotFound)
  File "/usr/local/lib/python3.12/site-packages/podman/api/client.py", line 74, in raise_for_status
    raise not_found(cause, response=self._response, explanation=message)
podman.errors.exceptions.ImageNotFound: 404 Client Error: Not Found (no such image: ghcr.io/openwrt/imagebuilder:ramips-mt7621-v24.10.1: image not known)

ERROR: Build failed with status 500
The above errors are often due to the upgrade server lagging behind the
build server, first suggestion is to wait a while and try again.

dc366, 2025-04-17:
Try again, it looks like there's an imagebuilder container for that target available now.

hipyskipy, 2025-04-17:
Thanks!  It's working now.

closed 2025-04-17 by hipyskipy