openwrt/asu #1354: Error building the firmware image Server response: init

d1stru3t0r, 2025-04-17:
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "99949c6ea39cb237950340f8a32d6f4d6319c767078777e4510db8155b90d303",
    "sha256_unsigned": "",
    "client": "luci/25.057.55135~802a31a",
    "packages": {
        "iwinfo": "2024.10.20~b94f066e-r1",
        "libopenssl": "3.0.16-r1",
        "luci-app-statistics": "25.057.55135~802a31a",
        "cache-domains-openssl": "2.4.0-r1",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "libpthread": "1.2.5-r4",
        "rpcd": "2024.09.17~9f4b86e7-r1",
        "libiwinfo": "2024.10.20~b94f066e-r1",
        "kmod-nft-fib": "6.6.73-r1",
        "kmod-nfnetlink": "6.6.73-r1",
        "kmod-nf-reject6": "6.6.73-r1",
        "kmod-crypto-aead": "6.6.73-r1",
        "luci-app-https-dns-proxy": "2023.12.26-r4",
        "luci-mod-system": "25.057.55135~802a31a",
        "kmod-nf-flow": "6.6.73-r1",
        "libnl-tiny": "2025.03.19~c0df580a-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "ucode-mod-uloop": "2025.02.10~a8a11aea-r1",
        "px5g-mbedtls": "11",
        "getrandom": "2024.04.26~85f10530-r1",
        "ucode-mod-ubus": "2025.02.10~a8a11aea-r1",
        "luci-theme-bootstrap": "25.057.55135~802a31a",
        "luci-app-irqbalance": "25.094.58090~02dde50",
        "kmod-pppoe": "6.6.73-r1",
        "kmod-pppox": "6.6.73-r1",
        "kmod-nf-reject": "6.6.73-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "kmod-nf-nat": "6.6.73-r1",
        "kmod-crypto-crc32c": "6.6.73-r1",
        "ucode-mod-uci": "2025.02.10~a8a11aea-r1",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "uboot-envtools": "2024.07-r1",
        "dnsmasq": "2.90-r4",
        "kmod-crypto-acompress": "6.6.73-r1",
        "rpcd-mod-ucode": "2024.09.17~9f4b86e7-r1",
        "ucode-mod-math": "2025.02.10~a8a11aea-r1",
        "kmod-lib-crc32c": "6.6.73-r1",
        "luci-mod-status": "25.057.55135~802a31a",
        "cache-domains-mbedtls": "2.4.0-r1",
        "kmod-nft-nat": "6.6.73-r1",
        "https-dns-proxy": "2023.12.26-r4",
        "kmod-usb3": "6.6.73-r1",
        "luci-app-firewall": "25.057.55135~802a31a",
        "libmbedtls": "3.6.3-r1",
        "libmnl": "1.0.5-r1",
        "jansson": "2.14-r3",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uclient-fetch": "2024.10.22~88ae8f20-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "ucode-mod-fs": "2025.02.10~a8a11aea-r1",
        "luci-ssl": "25.057.55135~802a31a",
        "dropbear": "2024.86-r1",
        "libnftnl": "1.2.8-r1",
        "rpcd-mod-file": "2024.09.17~9f4b86e7-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "ucode-mod-nl80211": "2025.02.10~a8a11aea-r1",
        "cache-domains-wolfssl": "2.4.0-r1",
        "kmod-crypto-authenc": "6.6.73-r1",
        "ucode": "2025.02.10~a8a11aea-r1",
        "kmod-nf-log": "6.6.73-r1",
        "luci-app-package-manager": "25.057.55135~802a31a",
        "urandom-seed": "3",
        "luci-proto-ppp": "25.057.55135~802a31a",
        "luci-mod-admin-full": "25.057.55135~802a31a",
        "ppp": "2.5.1-r1",
        "luci-base": "25.057.55135~802a31a",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "kmod-sched-mqprio": "6.6.73-r1",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-nf-log6": "6.6.73-r1",
        "luci-proto-ipv6": "25.057.55135~802a31a",
        "luci-app-attendedsysupgrade": "25.057.55135~802a31a",
        "kmod-crypto-des": "6.6.73-r1",
        "irqbalance": "1.9.4-r1",
        "kmod-crypto-hw-eip93": "6.6.73-r1",
        "luci-app-wifischedule": "25.057.55135~802a31a",
        "kmod-ppp": "6.6.73-r1",
        "kmod-nft-offload": "6.6.73-r1",
        "UDPspeeder": "20230206.0-r2",
        "uhttpd": "2023.06.25~34a8a74d-r4",
        "kmod-nf-conntrack": "6.6.73-r1",
        "kmod-mt7915-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "kmod-tcp-bbr": "6.6.73-r1",
        "ucode-mod-rtnl": "2025.02.10~a8a11aea-r1",
        "ca-bundle": "20240203-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "kmod-tls": "6.6.73-r1",
        "luci": "25.057.55135~802a31a",
        "luci-light": "25.057.55135~802a31a",
        "kmod-nf-conntrack6": "6.6.73-r1",
        "kmod-usb-ledtrig-usbport": "6.6.73-r1",
        "adblock-fast": "1.1.3-r5",
        "kmod-lib-lzo": "6.6.73-r1",
        "luci-app-adblock-fast": "1.1.2-r18",
        "rpcd-mod-iwinfo": "2024.09.17~9f4b86e7-r1",
        "luci-mod-network": "25.057.55135~802a31a",
        "kmod-nft-core": "6.6.73-r1",
        "libucode": "2025.02.10~a8a11aea-r1",
        "uhttpd-mod-ubus": "2023.06.25~34a8a74d-r4",
        "hostapd-common": "2024.09.15~5ace39b0-r2",
        "kmod-ipt-offload": "6.6.73-r1",
        "wireless-regdb": "2025.02.20-r1",
        "baresip-mod-plc": "1.1.0-r6",
        "urngd": "2023.11.01~44365eb1-r1",
        "kmod-slhc": "6.6.73-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "asus,rt-ax53u",
    "target": "ramips/mt7621",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "squashfs"
}

dc366, 2025-04-17:
Try again, it looks like the imagebuilder container is there now.

d1stru3t0r, 2025-04-17:
> Try again, it looks like the imagebuilder container is there now.

Now i have this:

STDERR:
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://downloads.openwrt.org/releases/24.10.1/targets/ramips/mt7621/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_core
Downloading http://downloads.openwrt.org/releases/24.10.1/targets/ramips/mt7621/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_base
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.1/targets/ramips/mt7621/kmods/6.6.86-1-2a6d68f3ded425efc0850296e7f62d53/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_kmods
Downloading http://downloads.openwrt.org/releases/24.10.1/targets/ramips/mt7621/kmods/6.6.86-1-2a6d68f3ded425efc0850296e7f62d53/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * check_data_file_clashes: Package libustream-openssl20201210 wants to install file /builder/build_dir/target-mipsel_24kc_musl/root-ramips/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-mbedtls20201210
 * opkg_install_cmd: Cannot install package cache-domains-openssl.
 * check_data_file_clashes: Package libustream-wolfssl20201210 wants to install file /builder/build_dir/target-mipsel_24kc_musl/root-ramips/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-mbedtls20201210
 * opkg_install_cmd: Cannot install package cache-domains-wolfssl.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2

d1stru3t0r, 2025-04-23:
Still no fix as of yesterday

Marssl78, 2025-04-26:
Error Building The Firmware Image
Server Response: init

Happens on 3 different AVM Fritz!Box 3370 Routers (target lantiq/xrx200) since more than a week now. 2 of them are micron flash-chip versions, one is hynix flash-chip.

{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.097.68136~9c86e03",
    "packages": {
        "luci-proto-wireguard": "25.097.68136~9c86e03",
        "iwinfo": "2024.10.20~b94f066e-r1",
        "libc": "1.2.5-r4",
        "opkg": "2024.10.16~38eccbb1-r1",
        "kmod-usb-dwc2": "6.6.73-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "ltq-vdsl-vr9-app": "4.17.18.6-r7",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "dnsmasq": "2.90-r4",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "kmod-ath9k": "6.6.73.6.12.6-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "dsl-vrx200-firmware-xdsl-a": "05.08.01.08.01.06.05.08.00.11.01.01-r1",
        "dropbear": "2024.86-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "kmod-ltq-ptm-vr9": "6.6.73-r4",
        "urandom-seed": "3",
        "ppp": "2.5.1-r1",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "luci-app-attendedsysupgrade": "25.097.68136~9c86e03",
        "kmod-ltq-atm-vr9": "6.6.73-r3",
        "luci-app-wol": "25.097.68136~9c86e03",
        "wpad-mbedtls": "2024.09.15~5ace39b0-r2",
        "kmod-ltq-vdsl-vr9": "6.6.73.4.17.18.6-r8",
        "ca-bundle": "20240203-r1",
        "luci": "25.097.68136~9c86e03",
        "fritz-tffs": "3",
        "kernel": "6.6.73~fe0372157644410cd19931638b8a5657-r1",
        "kmod-ltq-deu-vr9": "6.6.73-r45",
        "ltq-vdsl-vr9-vectoring-fw-installer": "6.8.6-r4",
        "dsl-vrx200-firmware-xdsl-b-patch": "05.08.01.08.01.06.05.08.00.11.01.01-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "ppp-mod-pppoa": "2.5.1-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "avm,fritz3370-rev2-micron",
    "target": "lantiq/xrx200",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "squashfs"
}

AustinZhang1024, 2025-06-24:
same problem now