openwrt/asu #1355: [TP-Link Archer AX23 v1] Error building the firmware image - Server response: init

hoshinetsu, 2025-04-17:
### Notes
This error keeps occuring when trying to upgrade a **TP-Link AX23** to **OpenWRT 24.10.1**.
Interestingly, a different device (**Asus TUF AX6000**) managed to upgrade effortlessly on the first attempt - few minutes before trying to update the **AX23** for the first time. Every subsequent attempt on the **TP-Link**, even after a 3 hour wait, has failed returning the same error message.
### Running firmware
- Model	TP-Link Archer AX23 v1
- Architecture	MediaTek MT7621 ver:1 eco:3
- Target Platform	ramips/mt7621
- Firmware Version	OpenWrt 24.10.0 r28427-6df0e3d02a / LuCI openwrt-24.10 branch 25.059.57589~8d7a815
- Kernel Version	6.6.73

### Error building the firmware image

Server response: init
```
Request Data:
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "1",
    "request_hash": "da03c60a4667d945751cf3b921310641e460f8a26b984004ebbe36cb990ab963",
    "sha256_unsigned": "",
    "client": "luci/25.059.57589~8d7a815",
    "packages": {
        "luci-proto-wireguard": "25.059.57589~8d7a815",
        "iwinfo": "2024.10.20~b94f066e-r1",
        "luci-app-statistics": "25.059.57589~8d7a815",
        "liblucihttp": "2023.03.15~9b5b683f-r1",
        "cgi-io": "2022.08.10~901b0f04-r21",
        "libc": "1.2.5-r4",
        "opkg": "2024.10.16~38eccbb1-r1",
        "wpad-basic-mbedtls": "2024.09.15~5ace39b0-r2",
        "nano": "8.3-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "ucode-mod-uloop": "2025.02.10~a8a11aea-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "kmod-pppoe": "6.6.73-r1",
        "btop": "1.4.0-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "dnsmasq": "2.90-r4",
        "kmod-crypto-acompress": "6.6.73-r1",
        "libmbedtls": "3.6.2-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "luci-ssl": "25.059.57589~8d7a815",
        "dropbear": "2024.86-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "luci-app-sqm": "25.059.57589~8d7a815",
        "urandom-seed": "3",
        "ppp": "2.5.1-r1",
        "kmod-leds-gpio": "6.6.73-r1",
        "kmod-gpio-button-hotplug": "6.6.73-r5",
        "logd": "2024.04.26~85f10530-r1",
        "luci-app-attendedsysupgrade": "25.059.57589~8d7a815",
        "kmod-ppp": "6.6.73-r1",
        "luci-app-wol": "25.072.79561~41a338d",
        "luci-app-upnp": "25.059.57589~8d7a815",
        "kmod-mt7915-firmware": "6.6.73.2025.01.14~8e4f72b6-r1",
        "htop": "3.3.0-r2",
        "ca-bundle": "20240203-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "luci": "25.059.57589~8d7a815",
        "kmod-lib-lzo": "6.6.73-r1",
        "kernel": "6.6.73~3abe85def815b59c6c75ac1f92135cb6-r1",
        "liblucihttp-ucode": "2023.03.15~9b5b683f-r1",
        "hostapd-common": "2024.09.15~5ace39b0-r2",
        "urngd": "2023.11.01~44365eb1-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "tplink,archer-ax23-v1",
    "target": "ramips/mt7621",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "squashfs"
}
```
Thank you for all the effort on maintaining the Project and your hard development work!

dc366, 2025-04-17:
Try again, it looks like an imagebuilder container was just built for that target.

hoshinetsu, 2025-04-17:
Now it worked. The firmware upgrade went flawlessly. All good as expected with Attended Sysupgrade.
Thank you for assistance!

closed 2025-04-17 by hoshinetsu