openwrt/asu #1358: Trying to Upgrade from 23.05.3-r23809-234f1a2efa to 23.05.05 using Attended Sysupgarde. Getting error as below

santomaharaj, 2025-04-18:
Error building the firmware image
Server response: Error: Impossible package selection

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23809-234f1a2efa",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "profile": "xiaomi,redmi-router-ax6000-stock",
    "target": "mediatek/filogic",
    "version": "23.05.5",
    "packages": [
        "attendedsysupgrade-common",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "eip197-mini-firmware",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-authenc",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-des",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-hw-safexcel",
        "kmod-crypto-manager",
        "kmod-crypto-md5",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha1",
        "kmod-crypto-sha256",
        "kmod-crypto-sha512",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-leds-ws2812b",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-libphy",
        "kmod-mac80211",
        "kmod-mt76-connac",
        "kmod-mt76-core",
        "kmod-mt7915e",
        "kmod-mt7986-firmware",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-phy-aquantia",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-thermal",
        "kmod-wireguard",
        "libblobmsg-json",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "liblucihttp",
        "liblucihttp-ucode",
        "libmbedtls",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-pbr",
        "luci-base",
        "luci-compat",
        "luci-lib-ipkg",
        "luci-light",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-proto-wireguard",
        "luci-ssl",
        "luci-theme-argon",
        "luci-theme-bootstrap",
        "mt7986-wo-firmware",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-mbedtls",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "rpcd-mod-ucode",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-html",
        "ucode-mod-math",
        "ucode-mod-nl80211",
        "ucode-mod-rtnl",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "ucode-mod-uloop",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-mbedtls"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-25.037.68351-2ca3f19"
}
STDERR:
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://downloads.openwrt.org/releases/23.05.5/targets/mediatek/filogic/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_core
Downloading http://downloads.openwrt.org/releases/23.05.5/targets/mediatek/filogic/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_base
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/23.05.5/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package luci-theme-argon.
make[2]: *** [Makefile:189: package_install] Error 255
make[1]: *** [Makefile:154: _call_manifest] Error 2
make: *** [Makefile:274: manifest] Error 2

dc366, 2025-04-18:
You have `luci-theme-argon` installed, which is a third party package, so it can't be part of an official generated image.

santomaharaj, 2025-04-18:
Thank you for the quick response. I was able to uninstall argon package and successfully complete the upgrade. Please go ahead and resolve/close this issue