openwrt/asu #1362: Response status 500: while downloading https://sysupgrade.openwrt.org/json/v1/overview.json

giovEra, 2025-04-18:
owut:
```
root@openwrt:~# owut download -v -V 24.10.1
owut - OpenWrt Upgrade Tool 2025.03.14~52e7d44c-r1 (/usr/bin/owut)
ERROR: Response status 500 while downloading
  https://sysupgrade.openwrt.org/json/v1/overview.json
```

efahl, 2025-04-18:
I believe this is a result of cache racing on the profiles, for which I just submitted a PR: https://github.com/openwrt/asu/pull/1363

Indificator, 2025-04-18:
[image]

kirula0626, 2025-04-18:
I am also facing the same error on my OpenWrt running on a Raspberry Pi 4B. @efahl 

[image]

sabbath88, 2025-04-18:
Seems to be fixed:

[image]

kirula0626, 2025-04-18:
@sabbath88 Thank you.

kirula0626, 2025-04-19:
@sabbath88 I think there is problem, Stuck in the queue on my side. what about you ??

sabbath88, 2025-04-19:
I stuck as well

closed 2025-04-21 by giovEra