openwrt/asu #1364: Stuck in queue?

FrozenGamesYT, 2025-04-18:
I have been trying to build update firmware but i have been stuck in queue at the same place for like 15-30 minutes?

[image]

warrenmr, 2025-04-19:
Same with me.  Queue position 251 on one device and 281 on the other.  45 ftw.

kirula0626, 2025-04-19:
Same here. I have been stuck same place for 6 hours.

dudu15051505, 2025-04-19:
1131 position, for 20h 

[image]

gazaret, 2025-04-19:
I've had same situation, but i try manual upgrading and it's work!
https://openwrt.org/docs/guide-user/installation/generic.sysupgrade#downloading_the_openwrt_upgrade_image

shAd09, 2025-04-20:
843 since yesterday

jvdpas, 2025-04-20:
Same problem here since 19-04-2025
[image]

amouzb, 2025-04-20:
[image]

opty77, 2025-04-20:
3281 and [stats](https://sysupgrade.openwrt.org/stats) doesn't work.

opty77, 2025-04-20:
[Forum](https://forum.openwrt.org/):

> 2025-April-19: There is a known issue that affects customized image building via the firmware selector and Attended Sysupgrade (ASU) / OpenWrt Upgrade Tool (OWUT). The system will queue build requests, but does not appear to be executing. As such, the queue has become very long and obviously nobody is able to obtain a customized image.
>
> Please do not create new threads about this issue since it is already known and being discussed [here](https://forum.openwrt.org/t/owut-openwrt-upgrade-tool/200035) and [here](https://forum.openwrt.org/t/the-openwrt-firmware-selector/81721).

[Thanks](https://forum.openwrt.org/t/the-openwrt-firmware-selector/81721/1341) @efahl for [stats](https://sysupgrade.openwrt.org/api/v1/stats)! (3368 now)

opty77, 2025-04-21:
@FrozenGamesYT: 0 now at [stats](https://sysupgrade.openwrt.org/api/v1/stats), close?

Gandulf78, 2025-04-26:
Same issue on my own ASU server running on Synology with Docker:

owut upgrade  --force
Password: 
ASU-Server     http://asu.xxx.synology.me
Upstream       https://downloads.openwrt.org
Target         ramips/mt7621
Profile        ubnt_edgerouter-x
Package-arch   mipsel_24kc
Version-from   24.10.1 r28597-0425664679 (kernel 6.6.86)
Version-to     24.10.1 r28597-0425664679 (kernel 6.6.86)
1 packages were downgraded
24 packages are out-of-date
Request hash:
  42294aebb7185be5c782546e18bb5fd4e9bd1f569ad8a28483bf78d0cf24909d
--
Status:   queued - 2 ahead of you
Progress:  11s total =  11s in queue +   0s in build

aparcar, 2025-04-27:
Strange, the worker process should pick up the job. Is it possible it's downloading the image builder but very slowly?

Gandulf78, 2025-04-27:
Two days after the status is still queued: {"status":202,"detail":"queued","queue_position":0,"enqueued_at":"2025-04-25T11:47:14.282159","request_hash":"157f28e598efc6d2d7631e23efecd5a600ecbb38e24f6c29242cb12bb994cc90"}

aparcar, 2025-04-27:
Like I said you don't have a worker to pick up the jobs, there is always a server and at least one worker to do the long running jobs. Please add the following to your compose file and open another issue if the problem persists. The initial issue discussed here is unrelated to your topic.

```yaml
  worker:
    image: "docker.io/openwrt/asu:latest"
    build:
      context: .
      dockerfile: Containerfile
    restart: unless-stopped
    command: rqworker --logging_level INFO
    env_file: .env
    environment:
      REDIS_URL: "redis://redis:6379/0"
    volumes:
      - $PUBLIC_PATH:$PUBLIC_PATH:rw
      - $CONTAINER_SOCKET_PATH:$CONTAINER_SOCKET_PATH:rw
    depends_on:
      - redis
  ```

closed 2025-04-27 by aparcar