openwrt/asu #1388: ASU server: access to xmlhttprequest has been blocked by cors policy

StanyoPeti, 2025-04-23:
I installed my own ASU server based on the guide here, but creating a new image doesn't work. I become the following error in the browser when I request an image:

Access to XMLHttpRequest at 'http://<server_IP>/api/v1/overview?1745428639689' from origin 'http://<router_IP>' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource.

GET http://<server_IP>/api/v1/overview?1745428639689 net::ERR_FAILED 301 (Moved Permanently)
(anonymous) | @ | luci.js?v=25.111.775…4aa1d-1745362806:59
(anonymous) | @ | luci.js?v=25.111.775…4aa1d-1745362806:40
Promise.then
request | @ | luci.js?v=25.111.775…4aa1d-1745362806:40
get | @ | luci.js?v=25.111.775…4aa1d-1745362806:64
handleCheck | @ | overview.js?v=25.111…4aa1d-1745362806:21
eval | @ | ui.js?v=25.111.77565…aa1d-1745362806:368

Uncaught (in promise) TypeError: Cannot read properties of undefined (reading 'ok')
 at eval (overview.js?v=25.111.77565~1b4aa1d-1745362806:21:1728)
eval | @ | overview.js?v=25.111…4aa1d-1745362806:21
Promise.then |  
handleCheck | @ | overview.js?v=25.111…4aa1d-1745362806:21
eval | @ | ui.js?v=25.111.77565…aa1d-1745362806:368

When I paste the 'http://<server_IP>/api/v1/overview?1745428639689' in the browser, then I become the result on the following url: 'http://<server_IP>/json/v1/overview.json'.

efahl, 2025-04-24:
Are you running a proxy between the server and your client?  `nginx` or `caddy`? 

See if this helps: https://github.com/openwrt/asu/issues/1002#issuecomment-2444432047

StanyoPeti, 2025-04-24:
I haven't used a proxy for this use case yet. Otherwise I use haproxy. I configure it and it works.
Thanks!

aparcar, 2025-04-24:
Yea I use caddy in front of the server. Feel free to add a Python version, I think it's a single package and one line of code for fastapi