openwrt/asu #1395: Attended Sysupgrade OpenWrt 24.10.0 to 24.10.1 on X86/64 in proxmox

ayubwaheed, 2025-04-30:
{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "24.10",
    "revision": "r28427-6df0e3d02a",
    "efi": null,
    "advanced_mode": "0",
    "request_hash": "ce47fd20b1138f697067836069e165418b3e31d0453c56c2f181942dd167122b",
    "sha256_unsigned": "",
    "client": "luci/25.116.36015~cca4654",
    "packages": {
        "luci-proto-wireguard": "25.116.36015~cca4654",
        "mkf2fs": "1.16.0-r3",
        "opkg": "2024.10.16~38eccbb1-r1",
        "libpthread": "1.2.5-r4",
        "kmod-ixgbe": "6.6.73-r1",
        "rpcd": "2024.09.17~9f4b86e7-r1",
        "libiwinfo": "2024.10.20~b94f066e-r1",
        "kmod-nft-fib": "6.6.73-r1",
        "kmod-nfnetlink": "6.6.73-r1",
        "kmod-nf-reject6": "6.6.73-r1",
        "nano": "8.4-r1",
        "luci-mod-system": "25.116.36015~cca4654",
        "kmod-nf-flow": "6.6.73-r1",
        "libnl-tiny": "2025.03.19~c0df580a-r1",
        "kmod-lib-crc-ccitt": "6.6.73-r1",
        "px5g-mbedtls": "11",
        "getrandom": "2024.04.26~85f10530-r1",
        "ucode-mod-ubus": "2025.02.10~a8a11aea-r1",
        "luci-theme-bootstrap": "25.116.36015~cca4654",
        "kmod-pppoe": "6.6.73-r1",
        "virtio-console-helper": "9.1.3-r1",
        "kmod-pppox": "6.6.73-r1",
        "kmod-nf-reject": "6.6.73-r1",
        "procd-ujail": "2024.12.22~42d39376-r1",
        "base-files": "1655~6df0e3d02a",
        "kmod-nf-nat": "6.6.73-r1",
        "kmod-crypto-crc32c": "6.6.73-r1",
        "ucode-mod-uci": "2025.02.10~a8a11aea-r1",
        "partx-utils": "2.40.2-r1",
        "libustream-mbedtls": "2024.07.28~99bd3d2b-r1",
        "libf2fs": "1.16.0-r3",
        "libsmartcols": "2.40.2-r1",
        "kmod-r8169": "6.6.73-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "dnsmasq": "2.90-r4",
        "rpcd-mod-ucode": "2024.09.17~9f4b86e7-r1",
        "kmod-crypto-acompress": "6.6.73-r1",
        "ucode-mod-math": "2025.02.10~a8a11aea-r1",
        "kmod-lib-crc32c": "6.6.73-r1",
        "luci-mod-status": "25.116.36015~cca4654",
        "kmod-fs-vfat": "6.6.73-r1",
        "kmod-nft-nat": "6.6.73-r1",
        "luci-app-firewall": "25.116.36015~cca4654",
        "grub2-efi": "2.12-r1",
        "libmbedtls": "3.6.3-r1",
        "libmnl": "1.0.5-r1",
        "jansson": "2.14-r3",
        "grub2-bios-setup": "2.12-r1",
        "odhcp6c": "2024.09.25~b6ae9ffa-r1",
        "uclient-fetch": "2024.10.22~88ae8f20-r1",
        "uci": "2025.01.20~16ff0bad-r1",
        "grub2": "2.12-r1",
        "ucode-mod-fs": "2025.02.10~a8a11aea-r1",
        "luci-ssl": "25.116.36015~cca4654",
        "dropbear": "2024.86-r1",
        "curl": "8.10.1-r1",
        "libnftnl": "1.2.8-r1",
        "rpcd-mod-file": "2024.09.17~9f4b86e7-r1",
        "mtd": "26",
        "odhcpd-ipv6only": "2024.05.08~a2988231-r1",
        "ucode": "2025.02.10~a8a11aea-r1",
        "kmod-nf-log": "6.6.73-r1",
        "luci-app-package-manager": "25.116.36015~cca4654",
        "urandom-seed": "3",
        "luci-proto-ppp": "25.116.36015~cca4654",
        "libcomerr": "1.47.0-r2",
        "ppp": "2.5.1-r1",
        "luci-mod-admin-full": "25.116.36015~cca4654",
        "luci-base": "25.116.36015~cca4654",
        "logd": "2024.04.26~85f10530-r1",
        "kmod-nf-log6": "6.6.73-r1",
        "librt": "1.2.5-r4",
        "kmod-tg3": "6.6.73-r1",
        "wireguard-tools": "1.0.20210914-r4",
        "remoteit": "5.4.0",
        "qemu-ga": "9.1.3-r1",
        "luci-proto-ipv6": "25.116.36015~cca4654",
        "luci-app-attendedsysupgrade": "25.116.36015~cca4654",
        "libext2fs": "1.47.0-r2",
        "e2fsprogs": "1.47.0-r2",
        "kmod-e1000": "6.6.73-r1",
        "kmod-nft-offload": "6.6.73-r1",
        "kmod-ppp": "6.6.73-r1",
        "libss": "1.47.0-r2",
        "uhttpd": "2023.06.25~34a8a74d-r4",
        "kmod-nf-conntrack": "6.6.73-r1",
        "ca-bundle": "20241223-r1",
        "kmod-forcedeth": "6.6.73-r1",
        "libuclient": "2024.10.22~88ae8f20-r1",
        "kmod-amd-xgbe": "6.6.73-r1",
        "kmod-dwmac-intel": "6.6.73-r1",
        "luci": "25.116.36015~cca4654",
        "kmod-nf-conntrack6": "6.6.73-r1",
        "luci-light": "25.116.36015~cca4654",
        "kmod-drm-i915": "6.6.73-r1",
        "kmod-lib-lzo": "6.6.73-r1",
        "luci-lib-uqr": "25.116.36015~cca4654",
        "kmod-amazon-ena": "6.6.73-r1",
        "kmod-e1000e": "6.6.73-r1",
        "rpcd-mod-iwinfo": "2024.09.17~9f4b86e7-r1",
        "luci-mod-network": "25.116.36015~cca4654",
        "kmod-nft-core": "6.6.73-r1",
        "libucode": "2025.02.10~a8a11aea-r1",
        "kmod-button-hotplug": "6.6.73-r3",
        "uhttpd-mod-ubus": "2023.06.25~34a8a74d-r4",
        "jsonfilter": "2025.04.18~8a86fb78-r1",
        "kmod-igb": "6.6.73-r1",
        "kmod-igc": "6.6.73-r1",
        "libe2p": "1.47.0-r2",
        "kmod-bnx2": "6.6.73-r1",
        "urngd": "2023.11.01~44365eb1-r1",
        "kmod-slhc": "6.6.73-r1",
        "ppp-mod-pppoe": "2.5.1-r1"
    },
    "profile": "qemu-standard-pc-i440fx-piix-1996",
    "target": "x86/64",
    "version": "24.10.0",
    "diff_packages": true,
    "filesystem": "ext4"
}
STDERR:
Generate local signing keys...
Generate local certificate...
Package list missing or not up-to-date, generating it.

Building package index...
Downloading http://downloads.openwrt.org/releases/24.10.1/targets/x86/64/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_core
Downloading http://downloads.openwrt.org/releases/24.10.1/targets/x86/64/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/x86_64/base/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_base
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/x86_64/base/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.1/targets/x86/64/kmods/6.6.86-1-af351158cfb5febf5155a3aa53785982/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_kmods
Downloading http://downloads.openwrt.org/releases/24.10.1/targets/x86/64/kmods/6.6.86-1-af351158cfb5febf5155a3aa53785982/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/x86_64/luci/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_luci
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/x86_64/luci/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/x86_64/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_packages
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/x86_64/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_routing
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/x86_64/routing/Packages.sig
Signature check passed.
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/openwrt_telephony
Downloading http://downloads.openwrt.org/releases/24.10.1/packages/x86_64/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package remoteit.
make[2]: *** [Makefile:234: package_install] Error 255
make[1]: *** [Makefile:171: _call_manifest] Error 2
make: *** [Makefile:349: manifest] Error 2

ayubwaheed, 2025-05-01:
> Where did this package come from? It doesn't appear to be in any of the
> OpenWrt feeds...

I have installed it to access my vm through remote.it via openwrt

efahl, 2025-05-01:
ASU server cannot build images with package outside the OpenWrt official feeds.  If you wish to use ASU for the upgrade, you'll have to remove that package, perform the upgrade and then reinstall the package after the system comes back up.

ayubwaheed, 2025-05-01:
Ok no problem, so upgrading issue only cause of remote.it package? I can
remove it for upgrade process no issue