openwrt/asu #1401: Can't upgrade to 24.10.1

Gang945, 2025-05-08:
My router is xiaomi AX3600 . I encounterd a problem when I tried to upgrade my router from 23.05.4 to 24.10.1. See as below. There might be something wrong. I need help , thanks.

[image]

[image]

[image]

[image]

Gang945, 2025-05-08:
> That target was renamed between the two releases. Edit `/tmp/sysinfo/board_name`, replace its contents with `qualcommax/ipq807x` and try again.

I have tried to replace its content from "xiaomi,ax3600" to "qualcommax/ipq807x" ,but  still failed. Thanks.

efahl, 2025-05-09:
Oops, I misled you, I confused changing target and board_name.

Anyhow, change that back, /tmp/sysinfo/board_name should be as before
`xiami,ax3600`

In /etc/openwrt_release change
`DISTRIB_TARGET='qualcommax/ipq807x'`

And in /etc/os-release change
`OPENWRT_BOARD="qualcommax/ipq807x"`

and try again...

Gang945, 2025-05-11:
> Oops, I misled you, I confused changing target and board_name.
> 
> Anyhow, change that back, /tmp/sysinfo/board_name should be as before
> `xiaomi,ax3600`
> 
> In /etc/openwrt_release change
> `DISTRIB_TARGET='qualcommax/ipq807x'`
> 
> And in /etc/os-release change
> `OPENWRT_BOARD="qualcommax/ipq807x"`
> 
> and try again...
> 
Problem solved. Thank you!

closed 2025-05-11 by Gang945