openwrt/asu #1402: Error: Impossible package selection

teckglobal, 2025-05-16:
I have been running an Openwrt successfully in the Oracle Cloud for well over a year and with the occasional error updating this has been an ongoing error for well over 2 weeks now. Its usually resolved and the problem has always been that the update server was just having issues for whatever reason but this time I wanted to ask if the server is still having problems or is it something else ?

{
    "url": "https://sysupgrade.openwrt.org",
    "branch": "SNAPSHOT",
    "revision": "r29080-1bb9592fc7",
    "efi": {
        "path": "/sys/firmware/efi",
        "type": "directory",
        "size": 0,
        "mode": 16877,
        "atime": 1747411315,
        "mtime": 1747411315,
        "ctime": 1747411315,
        "inode": 4180,
        "uid": 0,
        "gid": 0
    },
    "advanced_mode": "1",
    "request_hash": "",
    "sha256_unsigned": "",
    "client": "luci/25.081.38700~bf48c56",
    "packages": {
        "apk-mbedtls": "3.0.0_pre20250208-r2",
        "base-files": "1656~1bb9592fc7",
        "blkid": "2.40.2-r1",
        "busybox": "1.37.0-r4",
        "ca-bundle": "20240203-r1",
        "ca-certificates": "20240203-r1",
        "collectd": "5.12.0-r53",
        "collectd-mod-bind": "5.12.0-r53",
        "collectd-mod-conntrack": "5.12.0-r53",
        "collectd-mod-contextswitch": "5.12.0-r53",
        "collectd-mod-cpu": "5.12.0-r53",
        "collectd-mod-cpufreq": "5.12.0-r53",
        "collectd-mod-df": "5.12.0-r53",
        "collectd-mod-memory": "5.12.0-r53",
        "collectd-mod-network": "5.12.0-r53",
        "collectd-mod-ping": "5.12.0-r53",
        "collectd-mod-protocols": "5.12.0-r53",
        "collectd-mod-rrdtool": "5.12.0-r53",
        "collectd-mod-tcpconns": "5.12.0-r53",
        "curl": "8.12.1-r2",
        "ddns-scripts-cloudflare": "2.8.2-r65",
        "dnsmasq-full": "2.90-r4",
        "dropbear": "2024.86-r1",
        "e2fsprogs": "1.47.2-r1",
        "firewall4": "2024.12.18~18fc0ead-r1",
        "fstools": "2024.12.02~49d36ba2-r1",
        "fwtool": "2019.11.12~8f7fe925-r1",
        "getrandom": "2024.04.26~85f10530-r1",
        "grub2-efi-arm": "2.12-r1",
        "htop": "3.4.0-r2",
        "ipset": "7.21-r1",
        "irqbalance": "1.9.4-r1",
        "jansson4": "2.14-r3",
        "jshn": "2024.03.29~eb9bcb64-r1",
        "jsonfilter": "2024.01.23~594cfa86-r1",
        "kernel": "6.6.83~5dd5abe923cc3f9474aaa5866b382589-r1",
        "kmod-amazon-ena": "6.6.83-r1",
        "kmod-atlantic": "6.6.83-r1",
        "kmod-bcmgenet": "6.6.83-r1",
        "kmod-crypto-acompress": "6.6.83-r1",
        "kmod-crypto-crc32c": "6.6.83-r1",
        "kmod-crypto-hash": "6.6.83-r1",
        "kmod-dwmac-imx": "6.6.83-r1",
        "kmod-dwmac-rockchip": "6.6.83-r1",
        "kmod-dwmac-sun8i": "6.6.83-r1",
        "kmod-e1000e": "6.6.83-r1",
        "kmod-fs-vfat": "6.6.83-r1",
[... 130 lines cut ...]
    "diff_packages": true,
    "filesystem": "ext4"
}

STDERR:

Generate local signing keys...
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
read EC key
writing EC key
WARNING: opening /builder/packages/packages.adb: No such file or directory
Package list missing or not up-to-date, generating it.

Building package index...
ERROR: unable to select packages:
  libblobmsg-json20240329 (no such package):
    required by: world[libblobmsg-json20240329]
  libjson-script20240329 (no such package):
    required by: world[libjson-script20240329]
  libubox20240329 (no such package):
    required by: world[libubox20240329]
make[2]: *** [Makefile:226: package_install] Error 3
make[1]: *** [Makefile:167: _call_manifest] Error 2
make: *** [Makefile:344: manifest] Error 2

efahl, 2025-05-17:
It's something else.  Those ABI-versioned packages were updated and the package name changed (e.g., libubox20240329  became libubox20241219 and so on), so when the LuCI app sends the old name to be `apk add`ed, apk can't find it.  This is a part of the apk transition that is not handled yet, so you have to "uninstall" those packages in the LuCI app before you submit the upgrade job.  Since the libraries are dependencies of other packages, the new version will be automatically installed, so it all should work out fine.

This will be an ongoing issue with using main snapshots until the missing features in apk are completed and propagated into the ASU clients; expect to see it again...

(If you want to use owut, here's the current workaround to the same problem: https://forum.openwrt.org/t/owut-openwrt-upgrade-tool/200035/668 )

Xoloth, 2025-05-18:
> It's something else. Those ABI-versioned packages were updated and the package name changed (e.g., libubox20240329 became libubox20241219 and so on), so when the LuCI app sends the old name to be [...]

Do NOT do that. it just removes those packages and doesnt install new ones so you are basically bricking your firmware.

efahl, 2025-05-18:
> Do NOT do that. it just removes those packages and doesnt install new ones so you are basically bricking your firmware.

That is incorrect.  Even if you could specify them by canonical name (which you cannot with apk in its present state), these packages are direct dependencies of top-level packages and removing them from the ASU build request will have no effect on the resulting image.

The simple fact is that the packages no longer exist, which is the root cause of this issue:
```
$ apk add libubox20240329
ERROR: unable to select packages:
  libubox20240329 (no such package)
```

Xoloth, 2025-05-18:
> That is incorrect. Even if you could specify them by canonical name (which you cannot with apk in its present state), these packages are direct dependencies of top-level packages and removing them from the ASU build request will have no effect on the resulting image.

Im sorry but you are wrong. Just uninstall libjson-script20240329 and tell me whether you are able to reach luci or ssh.

efahl, 2025-05-18:
> Im sorry but you are wrong. Just uninstall libjson-script20240329 and tell me whether you are able to reach luci or ssh.

Wait, are you deleting the package from your device with apk, or doing as I stated above "... "uninstall" those packages in the LuCI app before you submit the upgrade job"?

Again, removing packages from the request will have no effect on your device, it simply tells the ASU server to build an image with or without certain packages, and since these are dependencies, removal has no effect at all.  The LuCI app doesn't give you enough feedback when it processes requests, but if you do it in `owut` it's the first thing you see:
```
$ owut upgrade --remove libjson-script20241219
owut - OpenWrt Upgrade Tool 2025.05.12~8353c4e9-r1 (/usr/bin/owut)
WARNING: Package 'libjson-script20241219' is a dependency, removal will have no effect on the build
...
```