openwrt/asu #1403: https://sysupgrade.openwrt.org down?

remco-space, 2025-05-19:
Hey wonderful OPENWRT people,

Trying to do a owut susupgrade on my CPE210 V1 router running 24.10 and I get the following error:

root@OpenWrt:~# owut upgrade
ERROR: Failed to connect
ERROR: Response status (null) while downloading
  https://sysupgrade.openwrt.org/json/v1/overview.json

Also the website https://sysupgrade.openwrt.org shows that surprisingly few build were done during the weekend. 

The website said: "report issues here", so that is why I am writing this report here. Do close it, if this is not the right place.

jannoke, 2025-05-19:
I can second that - It's down.

```root@MeshKitchen:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.5 r24106-10cc5fcd00 on ramips/mt7621 (tplink,archer-ax23-v1)
No data available (61)
```

It does ping thou:
```C:\Users\me>ping sysupgrade.openwrt.org

Pinging asu-02.infra.openwrt.org [45.140.183.87] with 32 bytes of data:
Reply from 45.140.183.87: bytes=32 time=39ms TTL=50
Reply from 45.140.183.87: bytes=32 time=40ms TTL=50
Reply from 45.140.183.87: bytes=32 time=41ms TTL=50

Ping statistics for 45.140.183.87:
    Packets: Sent = 3, Received = 3, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 39ms, Maximum = 41ms, Average = 40ms
Control-C
^C
```

jannoke, 2025-05-19:
I was going to setup my own server, but asu github repo specifies [[ImmortalWrt](https://sysupgrade.kyarucloud.moe/)](https://sysupgrade.kyarucloud.moe/) as working server. 

So make and edit:
```
root@MeshKitchen:/etc/config$ cat attendedsysupgrade

config server 'server'
        option url 'https://sysupgrade.kyarucloud.moe'

config client 'client'
        option upgrade_packages '1'
        option auto_search '1'
        option advanced_mode '1'

```

and you can make the upgrade:
```
root@MeshKitchen:/etc/config$ auc
auc/0.3.2-1
Server:    https://sysupgrade.kyarucloud.moe
Running:   23.05.5 r24106-10cc5fcd00 on ramips/mt7621 (tplink,archer-ax23-v1)
Available: 23.05.4 r28061-399f9a1db3
Requesting package lists...
 libopenssl: 3.0.15-1 -> 3.0.16-1
 luci-app-opkg: git-24.148.43905-2891ca4 -> git-24.302.38724-3ded930
 luci-mod-system: git-24.067.01860-7a82b2f -> git-24.365.26965-7378b30
 luci-theme-bootstrap: git-24.086.46634-1ffe078 -> git-25.113.44797-b7f1756
 base-files: 1562-r24106-10cc5fcd00 -> 1721-r28061-399f9a1db3
 dnsmasq: 2.90-2 -> 2.90-3
 luci-mod-status: git-24.212.61237-b6da3f2 -> git-25.104.33767-417c453
 libmbedtls: 2.28.9-1 -> 2.28.10-1
 libopenssl-legacy: 3.0.15-1 -> 3.0.16-1
 luci-app-firewall: git-24.067.01746-69867db -> git-25.093.36456-7eaec9d
 luci-ssl: git-23.035.26083-7550ad6 -> git-24.272.29284-d386ad6
 odhcpd-ipv6only: 2023-10-24-d8118f6e-1 -> 2023-10-24-d8118f6e-2
 luci-proto-ppp: git-24.135.44542-f1ec9c2 -> git-24.272.29284-d386ad6
 luci-mod-admin-full: git-19.253.48496-3f93650 -> git-24.272.29284-d386ad6
 ppp: 2.4.9.git-2021-01-04-4 -> 2.4.9.git-2021-01-04-7
 luci-base: git-24.264.56413-c7a3562 -> git-25.120.37830-21c1b60
 luci-proto-ipv6: git-24.086.45108-51aee90 -> git-24.272.29284-d386ad6
 openwrt-keyring: 2022-03-25-62471e69-2 -> 2023-05-22-455b399d-3
 luci-app-attendedsysupgrade: git-23.339.51123-138595a -> git-25.120.37830-21c1b60
 libopenssl-conf: 3.0.15-1 -> 3.0.16-1
 ca-bundle: 20230311-1 -> 20240203-1
 luci: git-23.051.66410-a505bb1 -> git-24.272.29284-d386ad6
 luci-light: git-23.024.33244-34dee82 -> git-24.272.29284-d386ad6
 luci-mod-network: git-24.264.56960-63ba3cb -> git-25.120.37830-21c1b60
 wireless-regdb: 2024.07.04-1 -> 2024.10.07-1
 ppp-mod-pppoe: 2.4.9.git-2021-01-04-4 -> 2.4.9.git-2021-01-04-7
Are you sure you want to continue the upgrade process? [N/y]
```

Euphoreka7, 2025-05-19:
> and you can make the upgrade:
> 
> ```
> ...
>  wireless-regdb: 2024.07.04-1 -> 2024.10.07-1
>  ppp-mod-pppoe: 2.4.9.git-2021-01-04-4 -> 2.4.9.git-2021-01-04-7
> Are you sure you want to continue the upgrade process? [N/y]
> ```

not really possible
```
Updated list of available packages in /builder/build_dir/target-x86_64_musl/root-x86/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Pseudo file "dev" exists in source filesystem "/builder/build_dir/target-x86_64_musl/root-x86/dev".
Ignoring, exclude it (-e/-ef) to override.
make[3]: *** [/builder/include/image.mk:399: /builder/build_dir/target-x86_64_musl/linux-x86_64/root.ext4] Segmentation fault
make[3]: *** Deleting file '/builder/build_dir/target-x86_64_musl/linux-x86_64/root.ext4'
make[2]: *** [Makefile:268: build_image] Error 2
make[1]: *** [Makefile:163: _call_image] Error 2
make: *** [Makefile:337: image] Error 2
init
Bad message (74)
```

jannoke, 2025-05-19:
> > So make and edit:
> > ```
> > root@MeshKitchen:/etc/config$ cat attendedsysupgrade
> > 
> > config server 'server'
> >         option url 'https://sysupgrade.kyarucloud.moe'
> > 
> > 
> > 
> > config client 'client'
> >         option upgrade_packages '1'
> >         option auto_search '1'
> >         option advanced_mode '1'
> > ```
> 
> Maybe I'm just ignorant to how ASU functions but... wouldn't this generate an image with ImmortalWRT packages instead of the vanilla OpenWRT packages?

You are right. It did work thou for simple router I did have. So I hope i can float back to openwrt when comes back up.

sigurasg, 2025-05-21:
The server is now returning 502s to all requests, it seems:
```
root@OpenWrt:~# owut list
ERROR: Response status 502 while downloading
  https://sysupgrade.openwrt.org/json/v1/overview.json
root@OpenWrt:~# wget https://sysupgrade.openwrt.org/json/v1/overview.json
--2025-05-21 11:38:03--  https://sysupgrade.openwrt.org/json/v1/overview.json
Resolving sysupgrade.openwrt.org... 2001:678:6e1:1001:be24:11ff:fe23:4c6d, 45.140.183.87
Connecting to sysupgrade.openwrt.org|2001:678:6e1:1001:be24:11ff:fe23:4c6d|:443... connected.
HTTP request sent, awaiting response... 502 Bad Gateway
2025-05-21 11:38:05 ERROR 502: Bad Gateway.
```

generic-internet-user, 2025-05-23:
Still down:
```
$ curl https://sysupgrade.openwrt.org/json/v1/overview.json
curl: (28) Failed to connect to sysupgrade.openwrt.org port 443 after 135609 ms: Could not connect to server
```

Euphoreka7, 2025-05-25:
> Looks like it's back!

https://asu.aparcar.org/ - still not responding
https://sysupgrade.openwrt.org/ - Bad message (74)

xhemp, 2025-05-25:
> > Looks like it's back!
> 
> https://asu.aparcar.org/ - still not responding
> https://sysupgrade.openwrt.org/ - Bad message (74)

Worked for me and for others by the looks of it:

[image]

It might have stopped again, though.