openwrt/asu #1452: opening /builder/packages/packages.adb: No such file or directory

Soromeister, 2025-07-30:
[image]

List of packages:

```
apk ath10k-firmware-qca9887-ct-full-htt ath11k-firmware-ipq8074 ath11k-firmware-qcn9074 base-files ca-bundle coreutils curl dnsmasq dropbear e2fsprogs firewall4 fstools gre htop ipq-wifi-xiaomi_ax9000 jq kmod-ath10k-ct kmod-ath11k-ahb kmod-ath11k-pci kmod-fs-ext4 kmod-gpio-button-hotplug kmod-leds-gpio kmod-nft-offload kmod-phy-aquantia kmod-qca-nss-dp kmod-udptunnel6 kmod-usb-dwc3 kmod-usb-dwc3-qcom kmod-usb3 libc libgcc libustream-wolfssl libwolfsslcpu-crypto logd losetup luci luci-app-attendedsysupgrade luci-app-irqbalance mtd ncat-full netifd nftables odhcp6c odhcpd-ipv6only ppp ppp-mod-pppoe uboot-envtools uci uclient-fetch urandom-seed urngd wget-ssl wpad-wolfssl
```

Fails with the error:
```
Error: Impossible package selection
STDERR
Generate local signing keys...
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
WARNING: can't open config file: /builder/shared-workdir/build/staging_dir/host/etc/ssl/openssl.cnf
read EC key
writing EC key
WARNING: opening /builder/packages/packages.adb: No such file or directory
Package list missing or not up-to-date, generating it.

Building package index...
ERROR: unable to select packages:
  libustream-wolfssl (no such package):
    required by: world[libustream-wolfssl]
  libwolfsslcpu-crypto (no such package):
    required by: world[libwolfsslcpu-crypto]
  wpad-wolfssl (no such package):
    required by: world[wpad-wolfssl]
make[2]: *** [Makefile:227: package_install] Error 3
make[1]: *** [Makefile:167: _call_manifest] Error 2
make: *** [Makefile:345: manifest] Error 2
STDOUT
```
The last build that worked with these packages was `r30544-88c7803197`.

efahl, 2025-07-31:
The wolfssl package builds are currently failing, https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/base/.  You need to wait until they succeed again or switch to the mbedtls versions of those packages.

Soromeister, 2025-08-01:
Seems to be fixed for build `r30634-8873e26c68`.