openwrt/asu #1467: Error connecting to upgrade server

electro-nic, 2025-08-16:
Attended Sysupgrade
The attended sysupgrade service allows to upgrade vanilla and custom firmware images easily.

This is done by building a new firmware on demand via an online service.

Currently running: 24.10.2 - r28739-d9340319c6 

Error connecting to upgrade server
Could not reach API at "https://sysupgrade.openwrt.org/json/v1/overview.json". Please try again later.

Internal Server Error

NormSoftwareEng, 2025-08-16:
I found alternate servers listed on repository home page [https://github.com/openwrt/asu](https://github.com/openwrt/asu)

efahl, 2025-08-16:
A fix has been submitted.  https://github.com/openwrt/asu/pull/1466

efahl, 2025-08-16:
Plus, this will go away as soon as the 23.05.6 release has completed building, but that may be some days away...