openwrt/asu #1472: Image not found: ghcr.io/openwrt/imagebuilder:ath79-generic-v23.05.6

szpeter80, 2025-08-27:
**Currently running**: 23.05.5 - r24106-10cc5fcd00
**Model**	TP-Link Archer C7 v5
**Architecture**	Qualcomm Atheros QCA956X ver 1 rev 0
**Target Platform**	ath79/generic
**Firmware Version**	OpenWrt 23.05.5 r24106-10cc5fcd00 / LuCI openwrt-23.05 branch git-24.264.56413-c7a3562

The error message i got upon firmware upgrade:

Error building the firmware image

Server response: Error: Image not found: ghcr.io/openwrt/imagebuilder:ath79-generic-v23.05.6. If this version was just released, please try again in a few hours as it may take some time to become fully available.

The request:

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r24106-10cc5fcd00",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "23.05",
    "efi": null,
    "request_hash": "4a43cf6f1181e60ed16a257a0e751965628e7f02b9f869387b7d7d262a4256bf",
    "profile": "tplink,archer-c7-v5",
    "target": "ath79/generic",
    "version": "23.05.6",
    "packages": [
        "ath10k-firmware-qca988x-ct",
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-ath9k",
        "kmod-ath9k-common",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha512",
        "kmod-gpio-button-hotplug",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-random-core",
        "kmod-slhc",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb2",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "liblucihttp",
        "liblucihttp-ucode",
        "libmbedtls",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-base",
        "luci-light",
        "luci-lua-runtime",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-mbedtls",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "rpcd-mod-ucode",
        "swconfig",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-html",
        "ucode-mod-math",
        "ucode-mod-nl80211",
        "ucode-mod-rtnl",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "ucode-mod-uloop",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-mbedtls"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.339.51123-138595a"
}
```

szpeter80, 2025-08-27:
running `podman image pull ghcr.io/openwrt/imagebuilder:ath79-generic-v23.05.6` by hand gives

```
Trying to pull ghcr.io/openwrt/imagebuilder:ath79-generic-v23.05.6...
Error: initializing source docker://ghcr.io/openwrt/imagebuilder:ath79-generic-v23.05.6: reading manifest ath79-generic-v23.05.6 in ghcr.io/openwrt/imagebuilder: manifest unknown`
```

A pull without an image tag (implicit 'latest') is successfull, so i think an image tag is missing or maybe the relevant image is not pushed to ghcr. 

Can you please advise a workaround ?

efahl, 2025-08-29:
Tested and working as of https://github.com/openwrt/openwrt/issues/19885#issuecomment-3233126894

This one can be closed.

szpeter80, 2025-08-30:
I can confirm, the reported issue has been resolved, closing this. Thank you very much !

closed 2025-08-30 by szpeter80